meshtastic/firmware #2042: [Feature Request]: Allow users out of canned messages

garthvh, 2022-12-15:
### Platform

NRF52, ESP32

### Description

It would be nice if users with a rotary encoder had a way out without sending a message, either a cancel at the bottom of the message list that advances the screen, or a timeout that advances the screen.  Currently once you twist the encoder you are forced to send a message to get out.

caveman99, 2022-12-30:
the select screen shoud time out after 20 seconds. This is a bug

garthvh, 2023-01-26:
The timeout is working again @caveman99, would it be hard to add cancel to the wheel of messages or is that a big hassle?

jp-bennett, 2023-07-30:
I'd like to see a "cancel" option, too. Really feels like a missing UX element.

mverch67, 2023-08-08:
What about a long press on the button to cancel? Would be in line with the screen ui PR.

_Long (2 sec) user button press exits to default local device screen at any time_

closed 2024-02-17 by thebentern