meshtastic/firmware #2043: [Bug]: RAK4631 silent halt (repeatable)

D4rk4, 2022-12-16:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.0.8.aa19718-d

### Description

The Port is still available, but I got a connection timeout when reconnecting with meshtastic cli.

### Relevant log output

```Shell
16:49:52 4160 [Router] Should encrypt MQTT?: 0
16:49:52 4160 [Router] Expanding short PSK #1
16:49:52 4160 [Router] Using AES128 key!
16:49:52 4160 [Router] nRF52 encrypt fr=25054758, num=ceaabaa, numBytes=21!
16:49:52 4160 [Router] enqueuing for send (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted rxtime=1671209392 rxSNR=6.5 rxRSSI=1.53492e-154)
16:49:52 4160 [Router] txGood=648,rxGood=1127,rxBad=39
16:49:52 4160 [Router] rx_snr found. hop_limit:4 rx_snr:6.500000
16:49:52 4160 [Router] rx_snr found in packet. Setting tx delay:2562
16:49:52 4160 [Router] FIXME-update-db Sniffing packet
16:49:52 4160 [Router] Delivering rx packet (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxtime=1671209392 rxSNR=6.5 rxRSSI=1.53492e-154)
16:49:52 4160 [Router] Forwarding to phone (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxtime=1671209392 rxSNR=6.5 rxRSSI=1.53492e-154)
16:49:52 4160 [Router] Update DB node 0x25054758, rx_time=1671209392
16:49:52 4160 [Router] NOTE: tophone queue is full, discarding oldest
16:49:52 4160 [Router] Module 'routing' considered
16:49:54 4161 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=37, time 804 ms
16:49:54 4161 [RadioIf] Lora RX (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted rxSNR=6 rxRSSI=1.53492e-154)
16:49:54 4161 [RadioIf] AirTime - Packet received : 804ms
16:49:54 4161 [Router] Found existing packet record for fr=0x25054758,to=0xffffffff,id=0xceaabaa
16:49:54 4161 [Router] Found existing packet record for fr=0x25054758,to=0xffffffff,id=0xceaabaa
16:49:54 4161 [Router] Add packet record (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted rxSNR=6 rxRSSI=1.53492e-154)
16:49:54 4161 [Router] Ignoring incoming msg, because we've already seen it (id=0x0ceaabaa fr=0x58 to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted rxSNR=6 rxRSSI=1.53492e-154)
16:49:54 4161 [Router] Incoming message was filtered 0x25054758
```

garthvh, 2022-12-16:
The python CLI times out all the time, you need to up arrow and try again, I don't see any Indication that the device is having an issue here

D4rk4, 2022-12-16:
No data transferred to air, no heartbeat blinking -> firmware issue

D4rk4, 2022-12-17:
Look like duty cycle related.
I guess need to add some output to the console log related to this and also fix the heartbeat blinking.

closed 2022-12-30 by garthvh