meshtastic/firmware #2045: [Bug]: TBeam panics when middle button pressed long without display

rpsainio, 2022-12-16:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.4.7022807f-d

### Description

TBeam panics when middle button is pressed long like when adjusting the brightness and the display is missing. It does not matter if the TBeam is in LS or running

### Relevant log output

```Shell
20:06:14 255343 [Router] Incoming message was filtered 0xc460f074
20:06:19 255348 [Power] Battery: usbPower=0, isCharging=0, batMv=3628, batPct=10
20:06:19 255348 [PositionModule] Channel utilization is >25 percent. Skipping this opportunity to send.
20:06:33 255362 [PowerFSM] lsEnter begin, ls_secs=300
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x601f8000  PS      : 0x3ffc4178  A0      : 0x00000000  A1      : 0x80138d06  
A2      : 0x3ffbf67c  A3      : 0x3ffbf708  A4      : 0x3ffbe504  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ffbd600  A9      : 0x3ffbf67c  
A10     : 0x3ffbf708  A11     : 0x3ffbe504  A12     : 0x3f41a635  A13     : 0x40138d6c  
A14     : 0x000249f0  A15     : 0x00000002  SAR     : 0x0000000a  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3f408e94  LBEG    : 0x400928e8  LEND    : 0x400928fe  LCOUNT  : 0x00000000  

Backtrace:0x601f7ffd:0x80138d06 |<-CORRUPTED

Core  0 register dump:
PC      : 0x40207cc2  PS      : 0x00060835  A0      : 0x80138520  A1      : 0x3ffbd6c0  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x8009ab24  A5      : 0x3ffbd5e0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x8013805e  A9      : 0x3ffbd690  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009ab24  A13     : 0x3ffbd5c0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40207cbf:0x3ffbd6c00x4013851d:0x3ffbd6e0 0x400987b0:0x3ffbd700 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   459][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (479232/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (53 Bytes)
??:??:?? 0  /prefs/config.proto (80 Bytes)
??:??:?? 0  /prefs/db.proto (598 Bytes)
??:??:?? 0  /prefs/module.proto (22 Bytes)
??:??:?? 0  /static/.gitkeep (0 Bytes)
??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0  /static/icon.svg.gz (842 Bytes)
??:??:?? 0  /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0  /static/index.85831067.css.gz (15140 Bytes)
??:??:?? 0  /static/index.d4fbecc1.js.gz (410906 Bytes)
??:??:?? 0  /static/index.html.gz (507 Bytes)
??:??:?? 0  /static/robots.txt.gz (42 Bytes)
??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   764][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   765][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   769][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
??:??:?? 0 Warning: Failed to find AXP2101 power management
[   788][I][XPowersAXP2101.tpp:230] ~XPowersAXP2101(): ~XPowersAXP2101
[   788][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   792][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
??:??:?? 0 AXP192 PMU init succeeded, using AXP192 PMU
[   818][D][XPowersAXP192.tpp:2010] setInterruptImpl(): setInterruptImpl DISABLE - 0xffffffffff

??:??:?? 0 =======================================================================
??:??:?? 0 DC1  : +   Voltage:3300 mV 
??:??:?? 0 DC2  : -   Voltage:1800 mV 
??:??:?? 0 DC3  : +   Voltage:3300 mV 
??:??:?? 0 LDO2 : +   Voltage:3300 mV 
??:??:?? 0 LDO3 : -   Voltage:3300 mV 
??:??:?? 0 =======================================================================
??:??:?? 0 Battery: usbPower=0, isCharging=0, batMv=3630, batPct=10
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 1 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=2.0.4.7022807f-d
??:??:?? 0 Setting random seed 2489531480
??:??:?? 0 Total heap: 196932
??:??:?? 0 Free heap: 163080
??:??:?? 0 Total PSRAM: 4192123
??:??:?? 0 Free PSRAM: 4191863
??:??:?? 0 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 OTA firmware version 0.1.1.98ab02e
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
```

caveman99, 2022-12-16:
2.0.4.7022807f-d is not an official build from our CI. I can not decode the stack trace without the ELF file from that build.

garthvh, 2022-12-16:
@caveman99 this does not seem to check for no screen https://github.com/meshtastic/firmware/pull/1856/files

rpsainio, 2022-12-16:
I am sorry, I forgot that I had this TBeam with special build. I will reproduce this with official one as it has been lurking around since 1.3-version

caveman99, 2022-12-16:
@garth indeed. still it's easier to fix when you have the stacktrace. And i seem to recall there was a fix for the crash on click&hold a few weeks back ...

caveman99, 2022-12-16:
> I am sorry, I forgot that I had this TBeam with special build. I will reproduce this with official one as it has been lurking around since 1.3-version

no problem, either an official build or the ELF file from your special build.

rpsainio, 2022-12-16:
@caveman99  click&hold sounds like close enough to my description and at least 2.0.7 did no longer react with crash. So, no hurry right now. I need own build to fix the problem of turning on/off GPS according to state of gps_enabled. I am testing battery times when GPS is turned off/on. I use TBeams with fixed positions.

garthvh, 2022-12-16:
> @garth indeed. still it's easier to fix when you have the stacktrace. And i seem to recall there was a fix for the crash on click&hold a few weeks back ...

This one https://github.com/meshtastic/firmware/issues/1646

rpsainio, 2022-12-17:
Just updated my build to 2.0.8.72 and the problem does not occur any longer

closed 2022-12-17 by rpsainio

caveman99, 2022-12-17:
@rpsainio the GPS rail will be switchable on the T-Beam soon. We have an incoming PR that will hopefully go into the next release.

rpsainio, 2022-12-17:
@caveman99 I hope the GPS-rail fix will fix this one as well #1951