meshtastic/firmware #2054: [Bug]: RAK 11200 crash when try access to HTTP

D4rk4, 2022-12-21:
### Category

Other

### Hardware

Rak11200

### Firmware Version

2.0.7.91ff7b9-d

### Description

Meshtastic-Web crash RAK12000, when file not found

### Relevant log output

```Shell
[ 51094][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html does not exist, no permits for creation
[ 51101][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html.gz does not exist, no permits for creation
WARNING file:stream_interface.py __reader line:179 Meshtastic serial port disconnected, disconnecting... device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

caveman99, 2022-12-21:
Is there more in this log? Usually a crash has some more output

D4rk4, 2022-12-21:
No, just reboot, prev messages are not related to the error and are always different.

garthvh, 2022-12-21:
> No, just reboot, prev messages are not related to the error and are always different.

Which base board are you using?

D4rk4, 2022-12-22:
RAK12000 + RAK19007 + RAK12002 + RAK1906

garthvh, 2022-12-22:
How did you flash the device?

D4rk4, 2022-12-22:
@garthvh esptool via device-install script =)

closed 2023-12-02 by thebentern