meshtastic/firmware #2059: [Bug]: Meshtastic burn screen on T-Beam!!!

D4rk4, 2022-12-22:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.0.8

### Description

After 5-10m after flash 2.0.8. display stops working. Just not showing anything, firmware still sees Screen (regarding log).

A downgrade to 2.0.7 did not help. Look like destroyed on the hardware level.

D4rk4, 2022-12-22:
Display - SPD0301

caveman99, 2022-12-22:
It's physically not possible to destroy these kind of displays through the firmware. Wear over months can ruin pixels, cracked edges of the screen may damage. But not within minutes and not through a software bug.

D4rk4, 2022-12-22:
@caveman99 I will try with another T-Beam

D4rk4, 2022-12-22:
After test with second unit got it:
[image]
Roll back firmware to 2.0.7 resolve issue.

caveman99, 2022-12-23:
https://github.com/meshtastic/firmware/compare/v2.0.7.91ff7b9...v2.0.8.090e166?expand=1

I have been through this diff up and down and can't for the life of me find a spot that will induce the described behaviour. Maybe you have more luck than me. Plus i don't have that display type here to counter-check :-(

D4rk4, 2022-12-23:
I will try with all artifacts related to this diff on weekend.

D4rk4, 2022-12-27:
PR2051 broke the display.

garthvh, 2022-12-27:
Have you tried setting the oled type under display settings?

D4rk4, 2022-12-27:
@garthvh nope, may you describe which params and values I need to provide to meshtastic cli.

garthvh, 2022-12-27:
Not totally sure, you can do the update from both android and iOS 

https://meshtastic.org/docs/settings/config/display#oled-defintion

garthvh, 2022-12-27:
Looks like it is display.oled in the protobufs

garthvh, 2022-12-28:
meshtastic --set display.oled OLED_SH1106 is probably what you need, I don't think any of the 1.3" screens are actually a SSD13XX

D4rk4, 2022-12-28:
@garthvh both OLED_SSD1306 and OLED_SH1106 not working for my SSD1309

garthvh, 2022-12-28:
Do you have a single screen like this?  They don't seem to have a good library and are 3X the cost of a 1.3" SH1106.

D4rk4, 2022-12-29:
@garthvh How it's related to costs? Also it's 1.51"

garthvh, 2022-12-29:
We have a limited amount of resources and nobody working on the firmware has this screen, at $30 few users are buying this screen to use with meshtastic. If you want support added for this screen it is going to be you adding it at this point as you are the only user who can test it

caveman99, 2022-12-30:
> PR2051 broke the display.

that is a huge rollup PR. Nevertheless i just went through it and again i can not find a place that would really impact the operation of this OLED screen.

D4rk4, 2022-12-31:
@caveman99 may you build firmware from PR2051 with all commits except changes related to Screen.* files? I hope it's related to dual-color screen support changes.

caveman99, 2022-12-31:
The 2.0.10 firmware can disable these changes interactively, in fact the default is now the old visual before 2.0.7

D4rk4, 2023-01-02:
@caveman99 how I can disable this in 2.0.10?

D4rk4, 2023-01-03:
@caveman99 is not actual, because firmware killed the second display.

D4rk4, 2023-01-07:
@caveman99 @garthvh Working with default settings on 2.0.12

thebentern, 2023-01-07:
Fixed. Closing

closed 2023-01-07 by thebentern