meshtastic/firmware #2065: [Board]: Happymodel ES900TX

lesykm, 2022-12-25:
### SOC

ESP32

### Lora IC

SX1276

### Product Link

https://www.happymodel.cn/index.php/2021/02/19/expresslrs-module-es915tx-long-range-915mhz-transmitter-and-es915rx-receiver/

### Description

This model has onboard PA which can produce up to 2W radio output, price is pretty cheap for this power.
Disadvantages - no I2C pins easily accessible, no extra serial pins easily accessible.
However in the future one make use of ES8285 neighbor chip.

garthvh, 2022-12-25:
Looks like a 82xx mcu which is not supportable

lesykm, 2022-12-25:
I'm sending messages to my mesh from this board right now :)

lesykm, 2022-12-25:
Oh yeah, on the board there are 2 chips
1. ESP32 - main one.
2. ESP8285 - "co-pilot". In Express LRS it is used for wifi-to-serial bridget, etc. It is not used in Meshtasctic (at least now).
Everything is controlled by jumpers, see this schematics: https://www.expresslrs.org/2.0/assets/images/Jumper-es900tx.png

garthvh, 2022-12-25:
> I'm sending messages to my mesh from this board right now :)
> 
RAM is tiny on the 82xx max nodes is going to be real low which is an issue.

lesykm, 2022-12-25:
I'm not using 82xx chip here, I'm using ESP32.

lesykm, 2022-12-25:
Boot log: https://pastebin.com/nwVe1HAE

caveman99, 2022-12-30:
Device log looks fine, do you still have unfinished business about that? :-) The module seems like a good PA option at 25 Euro a set. The PR needs a mod to src/platform/esp32/architecture.h to register as a regular board once we get a protobuf PR merged.

sachaw, 2023-01-26:
@lesykm Are you still working on this or need any assistance?

thebentern, 2023-05-23:
Interest in this device seems to have waned. Closing for now

closed 2023-05-23 by thebentern

xznhj8129, 2024-05-10:
This would be a pretty good device to support, it's become pretty popular since 2022.