meshtastic/firmware #2076: [Feature Request]:  Battery management system considerations

SR7673, 2022-12-29:
### Platform

ESP32

### Description

Talking about a solar powered system charging a Li-Ion battery, operating 24/7/365:
For the maximization of the battery live time it would be useful not to charge the battery up to 4.2 V in summer time. There will be plenty of energy available anyway, and fully charged batteries suffer in the heat. So it would be useful just to charge them up to say 3.9 V instead. However in winter, energy is rare, so any available mAh should be saved, thus it is useful to charge up to 100% or 4.20 V.
So, maybe it is possible to add a parameter to the meshtasic CLI that defines the upper charging voltage (default = 4.20). A command that can be transferred over the admin channel to a remote device. 
Does the ESP32 has an internal CPU temperature sensor so there can be a guess of the ambient temperature as well? (like the Raspi does!) That would allow to automatically regulate the uppe charging voltage based on the temperature.
Maybe a GPIO pin can be set HIGH during the charging period. This allows skilled people to use a MOSFET to switch the solar panel onto the battery (maybe on the RAK devices)...
Room for thoughts...

caveman99, 2022-12-29:
Not bad, but out of the scope of the firmware. The T-Beam is the only device that has a programmable charge controller, all other have a dumb charge IC or none at all. However there is an addon board developed by a discord member based on the telemetry module, containing an intelligent charge controller and temperature monitor.

Also, i have to admit i stopped reading after "ESP32" and "Solar Powered System". These are not a great fit and you'd be better off with nRF52 based systems like WisBlocks.

closed 2022-12-29 by caveman99

SR7673, 2022-12-29:
At least my nodes (1x T-Beam V1.1 and 2x TTGO Lora V2.1-1.6) show the battery voltage. So they must have an ADC on board. I hope the RAK4631 does have one too. That's all you need to realize a charge controller, together with a GPIO output that switches a MOSFET (turns the solar module on/off).

garthvh, 2022-12-29:
There is just no good reason to use an ESP32 for solar when a NRF device uses a fraction of the power draw without having to buy a bunch of extra hardware.

SR7673, 2022-12-29:
Yes but that's not what i wanted to discuss in this issue. If you prefer, imagine a RAK4631, running on a Li-Ion battery. This battery will have the same problem in summer time. It is simply not ideal to charge (and keep the voltage at that level until sunset) a Li-Ion cell up to 100% in summer time, with 40 °C ambient temperature.  
Using e.g. a T-Beam instead just consumes more energy per day, so you will need a larger solar module and more battery capacity but the problem with the heat in summer time at 100% charging voltage is just the same!
Thus: It would be a benefit to be able to vary the maximum charge voltage, maybe by a command that can be set remotely...

garthvh, 2022-12-29:
Only the tbeam has a chip that would enable you to set the charge current, it was in the protobufs power config in 1.2 and was removed.  More advanced battery management needs to be done outside of the main firmware.

SR7673, 2022-12-29:
As long as there is an onboard ADC that measures the actual battery voltage and as long as a GPIO pin is available, that can be used to switch a MOSFET, all you would need is the (external) MOSFET, nothing more.
My TTGO Lora V2.1-1.6 (i.e. NOT the T-Beam) shows the battery voltage. So it must have that ADC as well... I still don't know about the RAK4631 hardware. But 3 items are already ordered!

garthvh, 2022-12-29:
Perfect you can manage everything with a second MCU or use the remote hardware feature.

SR7673, 2022-12-29:
According to https://docs.rakwireless.com/Product-Categories/WisBlock/RAK4631/Datasheet/#overview
it has analog inputs. So the battery voltage can be measured without additional hardware. 

Remote hardware feature? I guess that means manually turning on/off a GPIO pin. But that's still not an automatic regulation/limitation of the voltage...

I see, i have to try to re-activate my C++ skills and build that on my own. Thought it would be a benefit for others too...

SR7673, 2022-12-29:
PS: Thus: It is not out of scope of firmware at all.

caveman99, 2022-12-29:
The firmware has power readings if the board has an ADC. Most boards do. Still the firmware needs to cut the charge power externally. This is really pout of the scope of meshtatic, there's dedicated charge controller boards that do exactly what you want.

SR7673, 2022-12-29:
Probably, yes. But such boards will have their own permanent power consumption, which is most likely much higher than the 2 uA the RAK4631 has during sleep mode. It can easily be 200 uA. And it is not remotely configurable. As i said, all you need is a switchable GPIO pin connected to the Gate of a MOSFET. This will switch the solar module. No need for a board and/or MCU.
The voltage measurement, the GPIO switching can be done in firmware, together with some commands (enabling/disabling that function | defining the end-charge voltage (e.g. 3.85 V in summer / 4.20 V in winter)). It would not be necessary to measure the voltage permanently. Once in 15 minutes would be fine (could be another parameter to be set by commands, similar to something like meshtastic --set position.position_broadcast_secs 900