meshtastic/firmware #2078: [Bug]: Device resets when trying to send message

GaryR-12, 2022-12-30:
### Category

Other

### Hardware

T-Beam

### Firmware Version

6978

### Description

Sometimes when sending a message from the iphone app, the device resets.

garthvh, 2022-12-30:
What config options have you set?

GaryR-12, 2022-12-30:
couldnt find any docs on how to set the options so left them as default.
Region set to US
Modem set to Long Range Fast

IIMacGyverII, 2022-12-30:
I have seen this as well.  Has been random for me also on a T-Beam with US and all default settings and iOS and one on wifi.  I first saw it on 2.0.06 and have seen it all the way just a few minutes ago on 2.0.10.  I could not tell you how to reproduce but I will keep looking.

garthvh, 2022-12-30:
If you run meshtastic --noproto you will see the error in the serial log

caveman99, 2022-12-30:
What firmware version is 6978? Please make sure your app (which app?)  and device firmware are up to date.

GaryR-12, 2022-12-30:
The app reports firmware as 2.0.6

GaryR-12, 2022-12-30:
> If you run meshtastic --noproto you will see the error in the serial log

Looks like the log only goes back a few hours. I will capture the log right away next time.

GaryR-12, 2023-01-01:
> If you run meshtastic --noproto you will see the error in the serial log
I observed this issue in the field. Will the error still be in the log by the time I connect to a PC to run the --noproto?
How can I capture this log in windows?

GaryR-12, 2023-01-01:
Tried to update the firmware using the web flasher but just get, 502 Bad Gateway

GMart, 2023-01-07:
It's happening to me with firmware 2.0.10. Device is a Heltec Lora 32 V3. Android app is 2.0.11. I'll update to firmware 2.0.12 and see what happens.

Logs:
```??:??:?? 33 BLE authentication complete
??:??:?? 34 [RadioIf] Completed sending (id=0x256a3b8f fr=0x5c to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=10)
??:??:?? 35 From Radio onread
??:??:?? 35 To Radio onwrite
??:??:?? 35 Client wants config, nonce=2
??:??:?? 35 Starting API client config
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_MY_INFO
??:??:?? 36 encoding toPhone packet to phone variant=3, 56 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_NODEINFO
??:??:?? 36 Sending nodeinfo: num=0xfa436f5c, lastseen=0, id=!fa436f5c, name=HeltecV3
??:??:?? 36 encoding toPhone packet to phone variant=4, 67 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_NODEINFO
??:??:?? 36 Sending nodeinfo: num=0x6d088d58, lastseen=1672544375, id=!6d088d58, name=TTGO portable
??:??:?? 36 encoding toPhone packet to phone variant=4, 76 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_NODEINFO
??:??:?? 36 Sending nodeinfo: num=0xfa4f80c4, lastseen=1672941847, id=!fa4f80c4, name=HomeBase
??:??:?? 36 encoding toPhone packet to phone variant=4, 87 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_NODEINFO
??:??:?? 36 Done sending nodeinfos
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 9 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 36 From Radio onread
??:??:?? 36 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 36 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 37 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CHANNELS
??:??:?? 37 encoding toPhone packet to phone variant=10, 6 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 6 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 23 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 20 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 20 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 12 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
??:??:?? 37 encoding toPhone packet to phone variant=5, 14 bytes
??:??:?? 37 From Radio onread
??:??:?? 37 getFromRadio=STATE_SEND_CONFIG
[... 159 lines cut ...]
??:??:?? 3 SX126xInterface(cs=8, irq=14, rst=12, busy=13)
??:??:?? 3 Starting meshradio init...
??:??:?? 3 Radio freq=906.875, config.lora.frequency_offset=0.000
??:??:?? 3 Set radio: region=US, name=LongFast, config=0, ch=19, power=30
??:??:?? 3 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 3 Radio myRegion->numChannels: 104 x 250.000kHz
??:??:?? 3 Radio channel_num: 19
??:??:?? 3 Radio frequency: 906.875000
??:??:?? 3 Slot time: 42 msec
??:??:?? 3 Set radio: final power level=22
??:??:?? 3 SX126x init result 0
??:??:?? 3 Frequency set to 906.875000
??:??:?? 3 Bandwidth set to 250.000000
??:??:?? 3 Power output set to 22
??:??:?? 3 Current limit set to 140.000000
??:??:?? 3 Current limit set result 0
??:??:?? 3 SX1262 Radio init succeeded, using SX1262 radio
??:??:?? 3 Not using WIFI
??:??:?? 3 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 3 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:?? 3 PowerFSM init, USB power=1
??:??:?? 3 Enter state: BOOT
[  3742][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
??:??:?? 3 [Power] Battery: usbPower=0, isCharging=0, batMv=0, batPct=0
??:??:?? 3 [Power] Heap status: 235572/289160 bytes free (-21572), running 25 threads```

garthvh, 2023-01-10:
I am seeing this too

??:??:?? 78 Original length - 7 
??:??:?? 78 Compressed length - 6 
??:??:?? 78 Original message - Dolphin 
??:??:?? 78 Using compressed message.
??:??:?? 78 Using AES256 key!
??:??:?? 78 ESP32 crypt fr=fa806a30, num=8baaf34a, numBytes=11!
??:??:?? 78 enqueuing for send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted)
??:??:?? 78 txGood=2,rxGood=0,rxBad=0
??:??:?? 79 [RadioIf] Starting low level send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 79 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 690 ms
??:??:?? 79 [RadioIf] AirTime - Packet transmitted : 690ms
??:??:?? 79 [RadioIf] Completed sending (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 82 [Power] Battery: usbPower=0, isCharging=0, batMv=4070, batPct=86
??:??:?? 82 [Power] Heap status: 181592/287528 bytes free (-1108), running 25 threads
??:??:?? 86 [Router] Sending reliable retransmission fr=0x0,to=0xffffffff,id=0x8baaf34a, tries left=2
??:??:?? 86 [Router] Found existing packet record for fr=0x0,to=0xffffffff,id=0x8baaf34a
??:??:?? 86 [Router] Add packet record (id=0x8baaf34a fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
??:??:?? 86 [Router] Should encrypt MQTT?: 0
??:??:?? 86 [Router] Original length - 7 
??:??:?? 86 [Router] Compressed length - 6 
??:??:?? 86 [Router] Original message - Dolphin 
??:??:?? 86 [Router] Using compressed message.
??:??:?? 86 [Router] Using AES256 key!
??:??:?? 86 [Router] ESP32 crypt fr=fa806a30, num=8baaf34a, numBytes=11!
??:??:?? 86 [Router] enqueuing for send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted)
??:??:?? 86 [Router] txGood=3,rxGood=0,rxBad=0
??:??:?? 86 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 690 ms
??:??:?? 86 [Router] Setting next retransmission in 7392 msecs:  (id=0x8baaf34a fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
??:??:?? 86 [RadioIf] Starting low level send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 86 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 690 ms
??:??:?? 86 [RadioIf] AirTime - Packet transmitted : 690ms
??:??:?? 87 [RadioIf] Completed sending (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 93 [Router] Sending reliable retransmission fr=0x0,to=0xffffffff,id=0x8baaf34a, tries left=1
??:??:?? 93 [Router] Found existing packet record for fr=0x0,to=0xffffffff,id=0x8baaf34a
??:??:?? 93 [Router] Add packet record (id=0x8baaf34a fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
??:??:?? 93 [Router] Should encrypt MQTT?: 0
??:??:?? 93 [Router] Original length - 7 
??:??:?? 93 [Router] Compressed length - 6 
??:??:?? 93 [Router] Original message - Dolphin 
??:??:?? 93 [Router] Using compressed message.
??:??:?? 93 [Router] Using AES256 key!
??:??:?? 93 [Router] ESP32 crypt fr=fa806a30, num=8baaf34a, numBytes=11!
??:??:?? 93 [Router] enqueuing for send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted)
??:??:?? 93 [Router] txGood=4,rxGood=0,rxBad=0
??:??:?? 93 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 690 ms
??:??:?? 93 [Router] Setting next retransmission in 7392 msecs:  (id=0x8baaf34a fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
??:??:?? 94 [RadioIf] Starting low level send (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 94 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 690 ms
??:??:?? 94 [RadioIf] AirTime - Packet transmitted : 690ms
??:??:?? 94 [RadioIf] Completed sending (id=0x8baaf34a fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64)
??:??:?? 101 [Router] Reliable send failed, returning a nak for fr=0x0,to=0xffffffff,id=0x8baaf34a
??:??:?? 101 [Router] Alloc an err=5,to=0xfa806a30,idFrom=0x8baaf34a,id=0x3de72cbd
??:??:?? 101 [Router] Enqueued local (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] cancelSending id=0x8baaf34a, removed=0
??:??:?? 101 [Router] Rx someone rebroadcasting for us (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] didn't find pending packet
??:??:?? 101 [Router] Add packet record (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] handleReceived(REMOTE) (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] Module 'routing' wantsPacket=1
??:??:?? 101 [Router] Received routing from=0xfa806a30, id=0x3de72cbd, portnum=5, payloadlen=2
??:??:?? 101 [Router] Routing sniffing (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] Received a nak for 0x8baaf34a, stopping retransmissions
??:??:?? 101 [Router] FIXME-update-db Sniffing packet
??:??:?? 101 [Router] Delivering rx packet (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] Forwarding to phone (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 [Router] Update DB node 0xfa806a30, rx_time=0
[... cut ...]
??:??:?? 101 phone downloaded packet (id=0x3de72cbd fr=0x30 to=0x30, WantAck=0, HopLim=0 Ch=0x0 Portnum=5 requestId=8baaf34a priority=120)
??:??:?? 101 encoding toPhone packet to phone variant=2, 32 bytes
??:??:?? 101 From Radio onread
??:??:?? 102 [Power] Battery: usbPower=0, isCharging=0, batMv=4054, batPct=84
??:??:?? 102 [Power] Heap status: 181600/287528 bytes free (8), running 25 threads

I don't seem to be registered a full reboot in the logs, but I see the screen show the boot screen, does not happen every time

SR7673, 2023-01-17:
Could it be that the power supply is to weak? Do you also run a battery in parallel? Maybe worth to consider...
What happens if you GET a message? Does it crash too?

IIMacGyverII, 2023-01-17:
In my case I don't see anyway how the power supply is causing this problem.  I have seen this problem many times now across 4 t-beams.  Multiple times the device is on battery which is always very close to full charge but I've also seen it multiple times with a full battery and plugged in to USB power charging so at least the external power supply is not the problem.  Perhaps it's the internal power supply/voltage converter but not the external one.

Another note, while I have not had the time to set up logging, I have now also seen this problem while triple tapping the PGM button to send an ad-hoc location message.  I first noticed this on 2.12 but that is also the first time I have tried the 3x PGM button feature.  I will bet it will do it in previous versions.  And yes I'm sure I did not press the reset button by accident.

Third note, I've seen this on t-beams using the iOS app to do immediate setup including setting region to US after a firmware update and I've seen this using the flasher program in windows to set the region and never connecting it to any app after the firmware flash.  This was using the 3x PGM button feature.

Forth note to answer SR7673's question about crashing while getting/receiving a message, while I cannot confirm that receiving the message was the actual fault, I have seen t-beams reboot at the same time they would have been receiving a message I've sent from another t-beam.  This could have been a crash during the receive function or the ack function being sent back to the other t-beam making it a sending crash.

Fifth note, the only times I have witnessed this is when I've been actually looking/working at/on the t-beam.  I have not looked or tried to find out if my devices are rebooting on their own while I am not using them.  If we knew it did or did not do this while sending it's non-message packets while idle, this could help narrow down where to look.

IIMacGyverII, 2023-02-09:
I upgraded to 2.0.18 the other day and have been seeing this reboot much more often since the upgrade from 2.0.12. I also made 2 big changes to how my mesh functions on all of my modules when I upgraded.  Changed from long-fast to long-slow and changed number of hops from 3 to 5. I just updated to 2.0.19 on one of my modules and will check over the next few days to compare reboots.

garthvh, 2023-02-09:
Long slow works well with maybe 3 nodes, increasing hops is going to increase airtime substantially. Canyou post your complete config?

IIMacGyverII, 2023-02-09:
All default for 4 of my nodes except long slow and 5 hops.  One of my modules is the same but it is using wifi for connection instead of bluetooth.

I am testing long_slow at the moment and testing five hops as well.  It would be great to be able to handle/auto negotiate multiple modem presets as I have multiple long distance points meant to allow smaller networks at multiple sites.  The long runs could use the slower preset between themselves and then use a faster one to talk to the local modules but I understand how difficult that would be.

garthvh, 2023-02-09:
That is probably too many nodes for long slow.  You need to use MQTT to bridge networks with different presets.

IIMacGyverII, 2023-02-09:
Yeah I figured it wouldn't be ideal but it was part of my testing.  But if it is causing more reboots, it would be worth sharing.  Either way I am still seeing the reboots and would like to help.

It does seem to help range but I am not getting all my messages through.  I am trying to figure out if it's because of the increased airtime or the number of nodes I have which is why I increased the number of hops.

I am planning on testing MQTT soon but the wifi web page interface seems broken/incomplete at the moment and would be very nice to use for that test.

garthvh, 2023-02-09:
WebUI is totally unnecessary for MQTT, literally does nothing for MQTT beyond turning it on.

IIMacGyverII, 2023-02-09:
yes but I want to be able to remote administrate it without leaving an android device next to it with teamviewer.

IIMacGyverII, 2023-02-09:
While MQTT is great and I love that it's implemented, I am building this for the need to communicate without internet.

garthvh, 2023-02-09:
MQTT is the only way to bridge two modem presets. There is no need for team viewer, you can connect via TCP on android.  I don't think you have any need for multiple modem presets, you can use the simulator to verify this.  Also probably not real likely that 5 hops is helping you any either, or if it is only one node benefits and the rest are just creating extra traffic

IIMacGyverII, 2023-02-09:
I understand and agree.  I wanted to try 5 hops to see what it would do and learn more about how Meshtastic works.  Meshtastic is still being developed and I am thoroughly enjoying it and all the hard work the community is putting in to it. 

 I have remote network access at each of my sites and being able to remote in to a site and then access the node via the web interface, makes my testing/learning much easier.  With the web interface in its current state, my only alternative is to have an android device at the remote site with teamviewer running so the android device can connect to the node via bluetooth.  I am patient and know it takes time and appreciate that and the work involved.  I hope I am not coming off ungrateful or disrespectful.  

That being said I did not mean to derail this thread away from the reboot issue.

francisco-pereira, 2023-04-28:
It's also happening to me.

Firmware: 2.1.5
Device: Heltec Lora 32 V3
Android app version: 2.0.19

Here some logs from when it happens:

DEBUG | 10:46:28 128 PACKET FROM PHONE (id=0x5b0fcb3c fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1)
handleReceived(USER) (id=0x5b0fcb3c fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1 rxtime=1682678788)
Module 'text' wantsPacket=1
INFO  | 10:46:28 128 [Screen] Received text msg from=0x0, id=0x5b0fcb3c, msg=Makdkdn
INFO  | 10:46:28 128 Enter state: ON
DEBUG | 10:46:28 128 showing standard frames
DEBUG | 10:46:28 128 Showing 0 module frames
DEBUG | 10:46:28 128 Total frame count: 83
DEBUG | 10:46:28 128 Added modules.  numframes: 0
DEBUG | 10:46:28 128 Finished building frames. numframes: 3
DEBUG | 10:46:28 128 Module 'text' considered
DEBUG | 10:46:28 128 Module 'canned' wantsPacket=1
DEBUG | 10:46:28 128 Module 'canned' considered
DEBUG | 10:46:28 128 Module 'ExternalNotificationModule' wantsPacket=1
INFO  | 10:46:28 128 External Notification Module Disabled
DEBUG | 10:46:28 128 Module 'ExternalNotificationModule' considered
DEBUG | 10:46:28 128 Module 'StoreForward' wantsPacket=1
DEBUG | 10:46:28 128 [Screen] Module 'StoreForward' considered
DEBUG | 10:46:28 128 [Screen] Module 'routing' wantsPacket=1
INFO  | 10:46:28 128 Received routing from=0x0, id=0x5b0fcb3c, portnum=1, payloadlen=7
DEBUG | 10:46:28 128 Routing sniffing (id=0x5b0fcb3c fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1 rxtime=1682678788)
Module 'routing' considered
DEBUG | 10:46:28 128 (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=27, time 160 ms
DEBUG | 10:46:28 128 Setting next retransmission in 6332 msecs:  (id=0x5b0fcb3c fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1682678788)
Add packet record (id=0x5b0fcb3c fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1682678788)
Original length - 7 
DEBUG | 10:46:28 128 Compressed length - 6 
DEBUG | 10:46:28 128 Original message - Makdkdn 
DEBUG | 10:46:28 128 Using compressed message.
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (nimble_host) 
Core  0 register dump:
PC      : 0x40380dd6  PS      : 0x00060c36  A0      : 0x4037a23d  A1      : 0x3fcbcf00  
A2      : 0x3fcbd431  A3      : 0x3c140627  A4      : 0x00000003  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x0000000c  A8      : 0x00000001  A9      : 0x3fcbd431  
A10     : 0x00000020  A11     : 0x3fcbd54b  A12     : 0x00000020  A13     : 0x3fcbd56a  
A14     : 0x00000004  A15     : 0x000000a7  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x40380dd3:0x3fcbcf00 0x4037a23a:0x3fcbcfe0 0x4211f93a:0x3fcbd010 0x42125da6:0x3fcbd320 0x42125de2:0x3fcbd3b0 0x4205b836:0x3fcbd3f0 0x4200516e:0x3fcbd490 0x42010d54:0x3fcbd500 0x42029be9:0x3fcbd520 0x42010a62:0x3fcbd540 0x42010d11:0x3fcbd590 0x42016261:0x3fcbd5b0 0x420164d1:0x3fcbd7b0 0x420165d3:0x3fcbd7d0 0x42010dbb:0x3fcbd7f0 0x42015df6:0x3fcbd810 0x420166f2:0x3fcbd830 0x420126f4:0x3fcbd850 0x4201278d:0x3fcbd880 0x42014262:0x3fcbd8a0 0x42014375:0x3fcbd8c0 0x42029767:0x3fcbd970 0x4203cd7b:0x3fcbd9a0 0x42046d35:0x3fcbdc20 0x42046d6d:0x3fcbdc50 0x4203fc9d:0x3fcbdc90 0x4203fe7d:0x3fcbdcc0 0x42040a0d:0x3fcbdce0 0x4203f38d:0x3fcbdd10 0x42048fd7:0x3fcbdd40 0x42047ac0:0x3fcbdd70 0x42047acb:0x3fcbdd90 0x403756f6:0x3fcbddb0 0x4203d13e:0x3fcbddd0

ELF file SHA256: 533782fa3ba665e0

E (26149) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
???ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037817c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (345) esp_core_dump_flash: No core dump partition found!
E (345) esp_core_dump_flash: No core dump partition found!
WARNING file:mesh_interface.py _sendToRadio line:606 Not sending packet because protocol use is disabled by noProto
INFO  | ??:??:?? 0 

------------------------------------------------------------------------------------------------------

[... 98 lines cut ...]
A6      : 0x00000001  A7      : 0x3fcbd244  A8      : 0x00000000  A9      : 0x0000002c  
A10     : 0x3fcbd43d  A11     : 0x3fcbd202  A12     : 0x00000002  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x0000000c  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x40380dd1:0x3fcbcf20 0x4037a23a:0x3fcbd010 0x42125da6:0x3fcbd320 0x42125de2:0x3fcbd3b0 0x4205b836:0x3fcbd3f0 0x42005225:0x3fcbd490 0x42010d54:0x3fcbd500 0x42029be9:0x3fcbd520 0x42010a62:0x3fcbd540 0x42010d11:0x3fcbd590 0x42016261:0x3fcbd5b0 0x420164d1:0x3fcbd7b0 0x420165d3:0x3fcbd7d0 0x42010dbb:0x3fcbd7f0 0x42015df6:0x3fcbd810 0x420166f2:0x3fcbd830 0x420126f4:0x3fcbd850 0x4201278d:0x3fcbd880 0x42014262:0x3fcbd8a0 0x42014375:0x3fcbd8c0 0x42029767:0x3fcbd970 0x4203cd7b:0x3fcbd9a0 0x42046d35:0x3fcbdc20 0x42046d6d:0x3fcbdc50 0x4203fc9d:0x3fcbdc90 0x4203fe7d:0x3fcbdcc0 0x42040a0d:0x3fcbdce0 0x4203f38d:0x3fcbdd10 0x42048fd7:0x3fcbdd40 0x42047ac0:0x3fcbdd70 0x42047acb:0x3fcbdd90 0x403756f6:0x3fcbddb0 0x4203d13e:0x3fcbddd0

ELF file SHA256: 533782fa3ba665e0

E (18374) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[... cut ...]
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037817c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (345) esp_core_dump_flash: No core dump partition found!
E (345) esp_core_dump_flash: No core dump partition found!
WARNING file:mesh_interface.py _sendToRadio line:606 Not sending packet because protocol use is disabled by noProto
INFO  | ??:??:?? 0