meshtastic/firmware #2085: [Bug]: Heltec lora 32 v3 usb and battery indicators are wrong. They're working the other way around.

megabyte2010, 2022-12-31:
### Category

Other

### Hardware

Other

### Firmware Version

tested 2.0.8 and 2.0.10

### Description

When I connect the usb cable, the device says that he is working with his connected battery and I can read voltage. When I disconnect the cable, the device says that he is working with usb cable, and I can´t read battery voltage. The display only goes to sleep when connected to usb cable power. on batteries the display never sleeps.

### Relevant log output

```Shell
heltec lora v3 connected to usb cable and with a almost full battery connected.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x27c0
entry 0x403c98c0
E (385) esp_core_dump_flash: No core dump partition found!
E (385) esp_core_dump_flash: No core dump partition found!
��@??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (479232/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (53 Bytes)
??:??:?? 0  /prefs/config.proto (91 Bytes)
??:??:?? 0  /prefs/db.proto (662 Bytes)
??:??:?? 0  /static/.gitkeep (0 Bytes)
??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0  /static/icon.svg.gz (842 Bytes)
??:??:?? 0  /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0  /static/index.c409c94b.css.gz (15150 Bytes)
??:??:?? 0  /static/index.ed76384f.js.gz (411080 Bytes)
??:??:?? 0  /static/index.html.gz (507 Bytes)
??:??:?? 0  /static/robots.txt.gz (42 Bytes)
??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   627][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
??:??:?? 0 Using analog input 1 for battery level
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x3 subtype probed in 2 tries 
??:??:?? 0 ssd1306 display found
??:??:?? 1 1 I2C devices found
??:??:?? 1 Meshtastic hwvendor=43, swver=2.0.8.090e166-d
??:??:?? 1 Setting random seed 4196873900
??:??:?? 1 Total heap: 291080
??:??:?? 1 Free heap: 256596
??:??:?? 1 Total PSRAM: 0
??:??:?? 1 Free PSRAM: 0
??:??:?? 1 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
??:??:?? 1 Setup Preferences in Flash Storage
??:??:?? 1 Number of Device Reboots: 9
??:??:?? 1 OTA firmware version 0.2.0.bd9c1f0
??:??:?? 1 Initializing NodeDB
??:??:?? 1 Loading /prefs/db.proto
??:??:?? 1 Loaded saved devicestate version 20
??:??:?? 1 Loading /prefs/config.proto
??:??:?? 1 Loaded saved config version 20
[  1748][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
??:??:?? 1 No /prefs/module.proto preferences found
??:??:?? 1 Installing default ModuleConfig
??:??:?? 1 Loading /prefs/channels.proto
??:??:?? 1 Loaded saved channelFile version 20
[  1792][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
??:??:?? 1 No /oem/oem.proto preferences found
??:??:?? 1 Number of Device Reboots: 9
??:??:?? 1 Expanding short PSK #1
??:??:?? 1 Wanted region 3, using EU_868
??:??:?? 1 region=3, NODENUM=0xfa436f60, dbsize=11
??:??:?? 1 Saving /prefs/db.proto
??:??:?? 1 Read RTC time as 1
??:??:?? 1 Using MSL altitude model
??:??:?? 1 WANT GPS=1
??:??:?? 3 Warning: Failed to find UBlox & MTK GNSS Module
??:??:?? 3 GxGSA NOT available
??:??:?? 3 Codec2 disabled (AudioModule 0, Region EU_868, permitted 0)
??:??:?? 3 External Notification Module Disabled
??:??:?? 3 Turning on screen
??:??:?? 3 Module wants a UI Frame
??:??:?? 3 Module wants a UI Frame
??:??:?? 3 SX126xInterface(cs=8, irq=14, rst=12, busy=13)
??:??:?? 3 Starting meshradio init...
??:??:?? 3 Radio freq=869.525, config.lora.frequency_offset=0.000
??:??:?? 3 Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27
??:??:?? 3 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
??:??:?? 3 Radio myRegion->numChannels: 1 x 250.000kHz
??:??:?? 3 Radio channel_num: 0
??:??:?? 3 Radio frequency: 869.525024
??:??:?? 3 Slot time: 42 msec
??:??:?? 3 Set radio: final power level=22
??:??:?? 3 SX126x init result 0
??:??:?? 3 Frequency set to 869.525024
??:??:?? 3 Bandwidth set to 250.000000
??:??:?? 3 Power output set to 22
??:??:?? 3 Current limit set to 140.000000
??:??:?? 3 Current limit set result 0
??:??:?? 3 SX1262 Radio init succeeded, using SX1262 radio
??:??:?? 3 Not using WIFI
??:??:?? 3 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 3 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:?? 3 PowerFSM init, USB power=1
??:??:?? 3 Enter state: BOOT
[  3468][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
??:??:?? 3 [Power] Battery: usbPower=0, isCharging=0, batMv=0, batPct=0
??:??:?? 3 [Power] Heap status: 237388/289256 bytes free (-19988), running 25 threads
??:??:?? 3 [Screen] Screen: Started...
??:??:?? 3 [SerialModule] Serial Module Disabled
??:??:?? 3 [AudioModule] Audio Module Disabled
??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
??:??:?? 5 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=42, time 870 ms
??:??:?? 5 [RadioIf] Lora RX (id=0x2a6db306 fr=0xbc to=0xff, WantAck=0, HopLim=1 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=0.240655)
??:??:?? 5 [RadioIf] AirTime - Packet received : 870ms
??:??:?? 5 [Router] Add packet record (id=0x2a6db306 fr=0xbc to=0xff, WantAck=0, HopLim=1 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=0.240655)
??:??:?? 5 [Router] Using channel 0 (hash 0x8)
```

caveman99, 2023-01-03:
@thebentern i think you have one of those, can you have a look? That doesn't seem right ... 

`??:??:?? 3 [Power] Battery: usbPower=0, isCharging=0, batMv=0, batPct=0`

megabyte2010, 2023-01-04:
I don´t know if this helps..
http://community.heltec.cn/t/esp32-v3-read-battery-voltage/11902

Jasons81, 2023-01-09:
I just got a Heltec V3 and more or less noticed the same thing last night, that the battery level never leaves the 70s. Also it will show battery when it's running off of USB.

megabyte2010, 2023-01-11:
You can only read battery voltage when he is connected with cable. Without, it shows the plug icon in the app.

GMart, 2023-01-14:
I did some experimenting and it looks like a value of 5.57 for ADC override will work on the Heltec V3 to get the battery to report correctly. It will show ~98% when fully charged and removed from the charger, and it will show +U (USB powered) when plugged in. It may need to be adjusted as all resistors have tolerance. Maybe some others can correlate this and then we can have the default changed in variant.h.

megabyte2010, 2023-01-14:
That value it's working for me. I have a solar powered board and I configured wit that value 10 minutes ago, and now I can see the battery voltage. Another heltec v3 connected with usb show the voltage +u

0xb1ff, 2023-02-02:
Here are my observations with firmware version 2.0.17, using a Heltec v3 without any battery, just powered by USB:
- After booting, the screen initially shows the "plug" symbol.
- After several seconds, the screen shows the "battery" symbol and the text "Unpowered...". On the second screen, the battery voltage is shown as 4.12V (which corresponds with the voltage that I can measure with a multimeter at the terminals, where the battery would usually be connected) and the battery SOC is shown as 91%.

Here are some things that I noted when trying to find out what causes this behaviour:
- I've read through the relevant source code in PowerFSM.cpp, but to understand it a rough description of the state machine design would be helpful.
- PowerFSM.h says "See sw-design.md for documentation", but I couldn't find that file. 
- In Power.cpp, line 182 says that for Heltecs with no battery connected, the measured voltage is 2204. It looks like this does not correspond with the 4.12V that my board measures.

0xb1ff, 2023-02-13:
Here are my observations with the same Heltec v3 module, still using firmware version 2.0.17, WITH a battery:
- After charging the battery, with USB disconnected, the battery symbol shows all bars and the battery voltage is shown as 4.15V with 95% SOC -> This is ok, in my opinion.
- When connecting USB, the battery image is shown with what is probably lightning to show that the battery is charging. The battery voltage is shown as 4.23V and the SOC as "0% +U" -> Shouldn't this be 100%? 4.23 V is within the 1% output voltage tolerance of the TP4054 battery charger circuit. (The 1% tolerance is given in the TP4054 datasheet.)

I suppose that I could fiddle around with the value for ADC_MULTIPLIER in variants/heltec_v3/variants , which is set to 5.22 in version 2.0.17, but then I might have to calibrate it for each board. It might be better, if the firmware would hide the tolerances of the charger circuit from the user and interpret all voltages between 4.16 V and 4.24 V as 100%.

(I still want to find out why the voltage was 4.12 V without a battery. This might take some days.)

0xb1ff, 2023-02-13:
Just an idea: It might be better to print "FULL" on the screen instead of "100%", as this exact percentage value implies more precision than supported by the hardware.

0xb1ff, 2023-02-13:
One more idea: The firmware could learn the maximum voltage that is reached during charging, store it and use it to calculate percentage values.

megabyte2010, 2023-02-26:
A possible fix is changing the default gps pins from 34/12 to gpio 45/46. I did that on the heltec lora 32 v3 and now I can see the battery voltage. And I changed the adc to 5.47 to give a more accurate reading. I did the same to the wireless stick lite board but didn't worked. 2.0.22 firmware.

0xb1ff, 2023-02-27:
What do you mean with getting a more accurate reading by changing the ADC multiplier? Did you measure the actual battery voltage with a multimeter? Did it deviate from the value shown on the screen?
In my case, with the unchanged ADC multiplier and with a battery connected to the board, the actual battery voltage and the voltage shown on the screen were exactly the same. In my opinion, this means that the multiplier is correct and does not need to be changed.

megabyte2010, 2023-02-28:
Yes I measured the voltage. What I have in the oled is what is showing in the multimeter.

garthvh, 2023-03-03:
Closing as we have two measured ADC multipliers that match a multimeter.

closed 2023-03-03 by garthvh