meshtastic/firmware #2088: Install Issue

seydunatirej, 2023-01-03:
I want to install the platform, but after installing visual studio, when I import the project, there are problems. I don't understand why. I would be glad if you help

sachaw, 2023-01-03:
We need more than that to go off. What problems are you running into, what steps have you performed and what have you tried to remedy the situation?

seydunatirej, 2023-01-03:
I will send the screenshot of the errors I get when I run the code. Because of these the firmware doesn`t upload in the device. I will be very glad if I could solve this. I hope you could help me. Thank you.. 
[image]
[image]

sachaw, 2023-01-03:
Not seeing what the issue is? Does it build?

seydunatirej, 2023-01-04:
It`s staying like in the first picture and it is not building.

sachaw, 2023-01-04:
There's no build output or error in the first picture, please confirm the following
1. You have selected your desired environment (board) in pio
2. You have clicked build (not build and upload), to confirm that it successfully builds
3. You will either get a build success (highlighted in green) or a fail (highlighted in red).
Please report back on what it is and any associated errors.

seydunatirej, 2023-01-04:
I thank you for helpfully instructions. Because I didn't choosed environment board it didn't worked. With your help its going well now.!

closed 2023-01-04 by sachaw