meshtastic/firmware #2090: [Feature Request]: Firmware version as telemetry value

tb0hdan, 2023-01-03:
### Platform

ESP32

### Description

Hi,

We, Ukrainian Meshtastic Community have been having troubles with figuring out which firmware version is most stable (e.g. right now 2.0.7 seems to be the one, while 2.0.8+ are loosing messages and node lists). 
We would like to request configuration toggle that enables/disables sending firmware version as telemetry value and the value itself.
We intend on monitoring these values and notifying respective owners of imminent upgrades/downgrades.

Thanks

garthvh, 2023-01-03:
The firmware version is in device telemetry and can be requested via the admin channel.

I think this is likely to be your lost message issue https://github.com/meshtastic/firmware/pull/1998

caveman99, 2023-01-03:
yes, this is new with 2.0.8 - best disable that on routers

tb0hdan, 2023-01-03:
I'll read up on how admin channel works. Still, idea is to have firmware version sent with telemerty and passive monitoring.

caveman99, 2023-01-03:
we discussed adding this to nodedb a while back and didn't do it cause of the memory overhead in Ram. if this is only transient info on a packet that does not save to the device store it would be feasible.

caveman99, 2023-01-03:
the 2.0.8 "problem" you can most likely resolve with `meshtastic --set lora.override_duty_cycle true`

tb0hdan, 2023-01-06:
Ok, closing issue then.

closed 2023-01-06 by tb0hdan