meshtastic/firmware #2091: [Feature Request]: Add SSD1327 LCD support

Krezalis, 2023-01-03:
### Platform

ESP32

### Description

Add SSD1327 LCD support.
Resolution 128 * 128, 16 level of colors/

caveman99, 2023-01-07:
I ordered one of these a few weeks back, unfortunately the chinese seller sent wrong boards (SH1106, 128*128*1) and is not very cooperative about changing them for the right ones. Searching around i could not find a single reputable vendor that carries these at the moment, so they look to be another victim of the chip shortage. Feel Free to reopen if these are available again in quantity :-)

closed 2023-01-07 by caveman99

Krezalis, 2023-01-07:
[image]
Have one yellow display, it looks fantastic. I paid for it $15, but now I see it is not popular.