meshtastic/firmware #2107: [Feature Request]: Add LAN support LAN8720

Krezalis, 2023-01-07:
### Platform

ESP32

### Description

Add LAN support LAN8720
- DHCP support for Web Interface and MQTT client

caveman99, 2023-01-07:
This module needs a grand total of 9 (nine) GPIOs for wiring. I am curious where you wanna source those on the ESP32 with user button, I2C and the Lora module connected.

Not a Fan.

caveman99, 2023-01-07:
Reference https://github.com/flusflas/esp32-ethernet

Krezalis, 2023-01-07:
Yes, it uses too many GPIOs. Have a very nice board TTGO T-Internet-COM and it works very stably. This board can be used as a repeater node.

garthvh, 2023-01-08:
Same problem as the Ethernet adaptor once you try and add a Lora radio.

thebentern, 2023-01-18:
Closing this as not planned

petrkr, 2024-05-14:
why is this closed with 'not possible', while it is possible?

I just made LAN7820 + PoE with Lora radio and still have enough pins for i2c and uart

Only problem I have now is understand whole concept of firmware code to integrate it as 'Ethernet' Option in config. Now I hard-coded it to main.cpp as proof-of-concept that it works and it is possible to use even 9pins PHY.

If meshtastic will use ESP-IDF 5.x.x there will be possible to use SPI ethernets like w5500 and share SPI with LoRa radio, so there would be needed only 2 gpios

How they requested in issue aswell #2449 and they also got 'not enough pins' answer

[image]

[image]

[image]

petrkr, 2024-05-14:
Caveman told that...

Im answer for his question, which had here 'where you want source those pins'.

Then

Use all 9 pins for eth, you can not change them.

Then

32,33,5,34,35 for LoRa
2,16 for i2c
4,36 for uart
39 for battery (but who will use battery in lan/poe powered board)
0 for user button

And still have free 4 pins from JTAG 12,13,14 and 15

And if there is lack of GPIOs for anything else, still can lot of things connct on i2c, even uart

caveman99, 2024-05-14:
It's great that you have it working now, would you willing to provide a PR? I wroute the original comment 18 months ago under a completely different pretext, just having implemented support for ETH on the rak module and having myriads of people asking for similar support on the ESP32 which was not an easy lift at the time. And we really had more pressing problems. So if you have it working, by all means submit it. I won't object for sure :-)

petrkr, 2024-05-14:
I have no PR as I wrote at 1st, i do not read whole source code, so i somehow ugly wrote it into main.cpp.

It would be easier, if someone from developers will know what all is needed to ifdef to make it possible in esp32 too.

For lucky esp idf have lwip stack, which does not care if you have wifi cli, wifi ap, eth or even LTE modem (pppos)

All will create tcp/ip stack and all libraries like http server, sockets etc will use it out-of-box.

Do not need implement some 'device related packet things' like w5500 in 'tcp' mode. SDK will handle it on freertos lwip level.

Unfortunately I have lot of projects to do and if everyone ask me for PR, I will have to do them on full time and for free. But I have to work on regular job, that is why I made this 'quick' proof-of-concept. As It is possible.

Now I asking for help if you/developers/ can help with implement, because you will know where those things are, I have to use search and try understand what everything to ifdef to turn on network (regardless type) to enable api server, mqtt etc..

Now I had to also enable wifi interface just in order all of those network relates things to start up

closed 2024-05-14 by caveman99

petrkr, 2024-05-14:
I read briefly eth classes, but they are exactly that 're-impementarion' of clasic tcp stack to use those ethernet modules, so I can not use now 'enable ethernet' option, it would need some ifdef and for esp32 just do eth.begin()... nothing else about tcp related stuff.

For luck you already using SDK 4.x which have it like that.

Then is also possible to have 'enable gprs/edge/lte' connection and it is also something like pppos.begin() and point it to uart.

There is littlebit 'problem' that esp have 3 uarts, 

1 used for debug
2 mostly gps (but for static installations useless)
3 could be that modem

But if someone what some nextion display  there is no other available.

But still there are uart-i2c bridges (i know about software serial, but that you do not want use witj esp32 at all)

petrkr, 2024-05-14:
> I don't get it - you have no time, but you need it? And suggest another developer pick it up? Not a single person on this project is paid, and we all have our day jobs. I can say for me this support is no prioriy and i won't work on it. If you need pointers, look for the W5500 support in the RAK4631 platform. Commit code, commit a blueprint, make a compelling case why this is a good idea overall. That's the way to bring this project forward. Not pointing out how superior your knowledge is without sharing it.

I asking for help, not for you to do it.

And for my use-case (node on roof with ethernet) it is enough that ugly hack...

petrkr, 2024-05-14:
And I want share, but want share atleast something useble, for that I asked for help. Maybe in my not native English it was bad said.

Goal is collaboration, not everything done by you or me. Help each other to achieve esp32 with lan (phy, spi,...) support

I now spend week to makr proof-of-concept (mostly HW side, boards, connections, custom builds) and now I was just aking for help to understand code quicker than reading it from scratch.

That's all.

Regards Petr

petrkr, 2024-05-14:
Proof of concept code https://github.com/petrkr/meshtastic-firmware/commit/ca1dabfb8e5d6f2fb7388cfec3bb9095f9195117

mstsclev, 2025-11-12:
It's unfortunate that this feature was never implemented.
Ethernet support would be very valuable for repeater setups.
Is there any way to revive or continue development on this feature?

petrkr, 2025-11-12:
> It's unfortunate that this feature was never implemented. Ethernet support would be very valuable for repeater setups. Is there any way to revive or continue development on this feature?

I did proof of concept, meshtastic developers does not accepted it, so I'll not longer waste time if even small change, which I needed as prerequest, was not merged.

But meshtastic is dead here, we already switched to MeshCore as that network works much better.