meshtastic/firmware #2115: [Bug]: RAK4631 crash firmware on Duty cycle limit

D4rk4, 2023-01-09:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.0.12

### Description

Repeatable 100%.
If it matters: RAK4631 without RAK's baseboard (RAK4631 in Mini-PCIe USB adapter).

### Relevant log output

```Shell
WARN  | 15:46:02 2623 [Router] Duty cycle limit exceeded. Aborting send for now, you can send again in 16 minutes.
ERROR | 15:46:02 2623 [Router] Error=9, returning NAK and dropping packet.
ERROR | 15:46:02 2623 [Router] Alloc an err=4,to=0x25053ad4,idFrom=0x2886bbb3,id=0x1765f41
[11049.287675] usb 1-1.4: USB disconnect, device number 13
```

garthvh, 2023-01-09:
Do you have a backtrace?

D4rk4, 2023-01-09:
@garthvh only logs output. May I capture a backtrace via `meshtastic --noproto`?

garthvh, 2023-01-09:
Should be in the log if there was really a crash, is it possible the custom board is not providing enough power and browning out? 

This is where your error is happening https://github.com/meshtastic/firmware/blob/49172e416e4d8ea0af924ecc5c569130310c45ea/src/mesh/MeshModule.cpp#L55

caveman99, 2023-01-09:
the RAK does not produce a backtrace unfortunately... Error 9 is thrown in route.cpp 139 (abortSendAndNak), the following Error 4 (no interface) is a bug @GUVWAF already solved, at least i recall seeing a diff somewhere. 'Alloc an Error 4' comes from MeshModule Line 55 which allocs space for the error package for sendAckNak in RoutingModule. here's either a failure in `router->sendLocal(p);` (RoutingModule Line 41) or the following codepath. We need to add debug printing to find that one.

caveman99, 2023-01-09:
i see what may be the problem. @D4rk4 is this node connected to an app via BLE or is it a router?

caveman99, 2023-01-09:
the sendLocal either s4ends stuff to the app (handleReceived) or back out on the network by calling send(). And this will fail again cause of duty cycle limit. (Router.cpp Line 201)

GUVWAF, 2023-01-09:
> Error 9 is thrown in route.cpp 139 (abortSendAndNak), the following Error 4 (no interface) is a bug @GUVWAF already solved, at least i recall seeing a diff somewhere. 

That commit seems to be lost now on master somehow.

I can't reproduce this using a RAK4631 connected with BLE when setting the duty cycle manually to 0.1%. 

> DEBUG | ??:??:?? 90 [Router] Sending reliable retransmission fr=0x0,to=0xffffffff,id=0xdabcaaa8, tries left=2
> DEBUG | ??:??:?? 90 [Router] Found existing packet record for fr=0x0,to=0xffffffff,id=0xdabcaaa8
> DEBUG | ??:??:?? 90 [Router] Add packet record (id=0xdabcaaa8 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
> WARN  | ??:??:?? 90 [Router] Duty cycle limit exceeded. Aborting send for now, you can send again in 58 minutes.
> ERROR | ??:??:?? 90 [Router] Error=9, returning NAK and dropping packet.
> ERROR | ??:??:?? 90 [Router] Alloc an err=4,to=0xba4bf9d0,idFrom=0xdabcaaa8,id=0x5f67572c
> DEBUG | ??:??:?? 90 [Router] Enqueued local (id=0x5f67572c fr=0xd0 to=0xd0, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=dabcaaa8 priority=120)
> DEBUG | ??:??:?? 90 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=24, time 649 ms
> DEBUG | ??:??:?? 90 [Router] Setting next retransmission in 7310 msecs:  (id=0xdabcaaa8 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)

garthvh, 2023-01-09:
> That commit seems to be lost now on master somehow.
>
> I can't reproduce this using a RAK4631 connected with BLE when setting the duty cycle manually to 0.1%.

He is using a custom base board, so there could be a hardware design issue

geeksville, 2023-01-09:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/running-rak4631-without-the-rak19007/7021/13

D4rk4, 2023-01-09:
@caveman99 It's installed in PCEngines APU and I grab these logs via console.
@garthvh I'm sure it's not related to power because transmission spends ~ 160 ma, idle ~3 ma for this module, and TX working without any issues.

caveman99, 2023-01-10:
@D4rk4 the throttle notification is going somewhere... the reason i am asking if something is connected to a phone API is i wanna trace the way this package takes... if it is delivered via API endpoint, there should be no issue, if the board is trying to send it out on the mesh again there may be a race condition or loop. Just looking for more info ...

D4rk4, 2023-01-10:
@caveman99 phone never connected/configured for this node =)

GUVWAF, 2023-01-10:
@caveman99 I think it doesn’t matter whether it’s connected via a phone API or not. The problem is probably that when it’s a packet from another node you have to rebroadcast, it will indeed try to send a NAK over the mesh, which results in an endless loop. I will create a PR later to only send the NAK when the packet originated at the same node, because then it is going to be send via the API and not over the mesh.

caveman99, 2023-01-10:
> @caveman99 The problem is probably that when it’s a packet from another node you have to rebroadcast, it will indeed try to send a NAK over the mesh, which results in an endless loop.

That was my original idea, when i tried to trace the package through the firmware. There was a 3 way conditional, and the last option calling send again which sent me back to where i started. I wanted to rule out that this was indeed the local node, in which case that path was not the right one, instead going to hnadleReceived() and the PhoneQueue

D4rk4, 2023-01-10:
@GUVWAF look like fixed by your PR #2120