meshtastic/firmware #2117: [Bug]: mqtt hostnames can't always be <32 characters

bmidgley, 2023-01-10:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.6

### Description

Sometimes hostnames we get for stuff in the cloud is really long and not negotiable. But we're pegged at 31 characters currently.

D4rk4, 2023-01-10:
Some users in the Russian telegram chat mentioned this issue before.
As a result of this limitation, we can't use free MQTT servers with random domain names for example HiveMQ.

garthvh, 2023-01-15:
This has been bumped to 64

closed 2023-01-15 by garthvh

xhci, 2024-12-18:
@garthvh by default, my ActiveMQ hostname on AWS is 67 characters (not including the port). It requires significantly more cost like a network load balancer to use custom hostnames. Would it be possible to bump this higher?

garthvh, 2024-12-18:
> @garthvh by default, my ActiveMQ hostname on AWS is 67 characters (not including the port). It requires significantly more cost like a network load balancer to use custom hostnames. Would it be possible to bump this higher?
> 
> I added an MR here that will probably fix the issue: #5604

This can't be merged (and would have to be made in the protobufs repo) There is no space left on the MQTT config packet.

xhci, 2024-12-18:
Thanks for the feedback. I'll figure out a solution on my end.

bmidgley, 2025-02-07:
We made a short cname in our case to avoid the long cloud hostname