meshtastic/firmware #2122: TTGO Esp32 LoRa v1

mAsO1975, 2023-01-11:
### Category

Hardware Compatibility

### Hardware

T-Lora v1

### Firmware Version

firmware-tlora-v1-2.0.0.18ab874

### Description

Hi, my name is Miguel Angel, I'm new to Mestastic, I'm trying to flash version v2.0.0.18 to my TTGO LoRa32 v1, and it gives me the following error:

*****************************************************************************************************
Using 'COM6' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting......

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: C8:C9:A3:FB:FE:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
*****************************************************************************************************

The most recent version that I have been able to lshear is v1-1.2.65.0, and the android app tells me that the firmware version is old and I have to update it, so I have tried to update to v2.0.0.18, with the consequent error, I hope someone can help me.

Thank you, and greetings to the entire community.

caveman99, 2023-01-11:
How are you trying to flash those? the easiest way these days is the web flasher on https://flasher.meshtastic.org/

Also: are you really sure this s a v1? please compare images in our docs with the actual device on https://meshtastic.org/docs/supported-hardware

mAsO1975, 2023-01-11:
Hello, thanks for your answer, I am using ESPHome, the device I am using is the one in the image purchased on Aliexpress.

[image]

caveman99, 2023-01-11:
ESPHome expects the firmware to be at offset 0x10000, our firmware image starts at 0x0. Try flashing the 'update' firmware that is based of 0x10000

Supported firmware flashing methods are listed here. https://meshtastic.org/docs/getting-started/flashing-firmware - if you use something else, you are on your own :-)

closed 2023-01-11 by caveman99