meshtastic/firmware #2128: [Feature Request]: Transmission details for Serial module, AT commands?

bobricius, 2023-01-12:
### Platform

ESP32

### Description

Hi,
I found way how to integrate my messenger (Picomputer/Armachat) in to meshtastic network,
It is now work like dumb terminal, display/keyboard for T-Beam.
By serial module I can now send messages, but missing some details about transmission.
I think for Meshtastic missing something like simple serial AT commands.
For everything it need some app or python skript.
By implementing AT commands it can be more usable like etc modem.
Some people want eliminate / minimize mobile phone from messaging.
There is video from test message sending https://youtu.be/wA_hsSf8uiw

Thanks for excellent work!

Peter

caveman99, 2023-01-12:
That's a really nice use for the serial module. If your review the settings for that module, you will find that you can set it to different operationg modes. At the moment you are using textmsg. If you set the mode to 'proto' the full binary client API as documented [here](https://meshtastic.org/docs/development/device/client-api) will be available on the serial port. That's the same API used by the Android and iOS apps over Bluetooth.

I you include the nanopb and our protobuf definitions into armachat, you can fully remote control the connectd node, change settings and send/receive messages and metadata.

Unfortunately implementing an AT-Style command interface is out of the scope of this project, due to the event driven nature of the whole system, and because we already have a working interface :-) It's not exactly low hanging fruit, but once you got the initial want_config transfer processed you'll get the hang of it fast.

bobricius, 2023-01-12:
I've tried many times to understand how protobuf works, but I haven't succeeded yet. I know that it is used everywhere in Meshtastic. I have to find out if there is an implementation for circuitpython. Thank you for the answer.

garthvh, 2023-01-13:
Think of it as json without any brackets or colons, I got hung up on it for a while too but it is just a really lightweight data format.

caveman99, 2023-01-13:
i am not too savvy with Circuitpython, but can you pull stuff in from regular PyPi? https://pypi.org/project/nanopb/ should get you started.

closed 2023-02-11 by garthvh