meshtastic/firmware #2135: [Bug]: Git chaos removed at least 2 fixes for SerialModule

TilCreator, 2023-01-12:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

v2.0.10.e09b12c

### Description

The commit https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-d68fc7a0a96158febd5d87686deec70ecf54a3429c0ed339c91df3d4abbbb6a2L218 seams to have removed the fixes https://github.com/meshtastic/firmware/pull/2050 and #2058, because those were merged to the master and not develop branch. The #2050 fix is for sure missing from versions older than 2.0.10, but available in 2.0.8.
I don't know if other PRs were overwritten by https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b.

Please re-add or otherwise fix the lost changes.

TilCreator, 2023-01-13:
I tried finding some other things that might have been overwritten by 14be4ee:

#2050

#2058 

Non 2.4Ghz support checking
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-34d21af3c614ea3cee120df276c9c4ae95053830d7f1d3deaf009a4625409ad2L465

Godmode
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-aea103aeb7f885321af1a87fdca390f21ff89b880e490f4ab6052830ee6d9226L70
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-e66d9302ef6e0c408459705ccfda378b51c6ec2ac14ab3ff20c1402e2e3f622aL226
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-4252194205e90e15b5e9517c29c59c76a24660a6a1f5d05617fe3ec400eb1787L30

FIXME on "optimisation"
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-e129b9a43d6aa0cbf5069f17506cb6d3da14c6eb5340e3bd003875c1787faa95L40
- https://github.com/meshtastic/firmware/commit/14be4ee9f050645dd7e6a4bcd6cdb92e1ed8969b#diff-e129b9a43d6aa0cbf5069f17506cb6d3da14c6eb5340e3bd003875c1787faa95L57

caveman99, 2023-01-13:
- Part of the 2.4 GHz support was moved to another location, this is intentional. 
- Godmode part 1 was intentional, removing a NOOP
- Godmode Part 2&3 were intentional, not neccessary any more cause of changes in Radiolib upstream.
- The 2 optimization changes are right, they imßplement the new direct message Ack

From the 2 serial module fixes, only paerts were reverted from a parallel commit implementing the Protobuf mode. This may indeed be a manual merge error. I have gone through the complete history of merges and didn't find any more occurances of this.

Thanks for bringing that to our attention though.

closed 2023-01-14 by thebentern