meshtastic/firmware #2140: [Bug]: after update to 2.10 no gps data HELTEC V2

LO77ARO, 2023-01-15:
### Category

Other

### Hardware

Heltec v2

### Firmware Version

2.10

### Description

After upgrading to 2.10 I can't get the devices to transmit or share their gps location. I have tried with different phones, checked the permissions of the app and nothing. I have 5 modules, none of them work. If I assign it a fixed location, it transmits it. The app is the latest version downloaded from the playstore.

garthvh, 2023-01-15:
Do you have hardware gps units or are you sharing the location from the phone?

LO77ARO, 2023-01-15:
after a clean install of de 2.0.10 and a clear view of the sky, the gps fix, and all work as i should.

closed 2023-01-15 by LO77ARO