meshtastic/firmware #2141: [Feature Request]: Change "No GPS Module" when location is from app

GMart, 2023-01-15:
### Platform

NRF52, ESP32

### Description

When I use a board without GPS installed, and I'm using location from my phone, currently the last screen shows "No GPS Module". That is technically correct, however I think it would be more useful to show something like "Location from phone"/"Location from app" instead. With the way it is, you have no way of knowing if your location is being sent over the mesh by looking at the device. Maybe even show the coordinates being sent over the mesh.

The default behavior could be kept if the device is not receiving a location from the app/ not connected and doesn't have a GPS module.

GMart, 2023-02-26:
Has there been any update on this? Or has there been any changes in the past few versions of the firmware that would allow for this?

code8buster, 2023-03-01:
Working on it. Stay tuned, doing some testing before sending the pull request

closed 2024-03-28 by thebentern