meshtastic/firmware #2144: [Feature Request]: User configurable center frequency

code8buster, 2023-01-15:
### Platform

NRF52, ESP32

### Description

Tracking issue for user configured frequency

As is, there is no direct way to start the LoRa modem at a given frequency. All of it is arbitrated by the RadioInterface in terms of frequency ranges and regions.
We do not have a region or band plan for licensed amateur radio, and to make any would be onerous due to the variability of bands, national/regional usage differences, etc. The number of RDEFs would expand greatly.

For users who set is_licensed=1, it may be better to expose a method of setting frequency. (RadioInterface::saveFreq a good candidate?)

caveman99, 2023-01-15:
Another possibility would be a custom RDEF slot that can be set via lora config. But just speccing the frequency is a lot easier. power Region limitation is off for is_licensed anyway and i think we could also override the other settings (voice, duty cycle) if is_licensed is true.

faydr, 2023-01-15:
Just to advocate further for this feature -- if hams (such as myself) want to set up "infrastructure" Meshtastic nodes (e.g., fixed nodes on top of a mountain), we need to coordinate with other users in the band (like other repeaters). Readily knowing the exact frequency in this case is essential to coordinating with other repeater owners.

caveman99, 2023-01-17:
We're shuffling protobufs right now, ths is on hold till the transition is done.

mc-hamster, 2023-01-18:
May also need to consider other overrides, eg bandwidth &  coding since those won't have any specific region.

@caveman99 I'd advocate for a prescribed frequency / modem configuration. This won't be the last time this comes up.

mc-hamster, 2023-01-18:
... now that (I think) we have plumbing for the 2.4ghz lora radios, this may be used for a possible future Meshtastic EME bounce. Imagine the headline "Meshtastic goes to the moon"

caveman99, 2023-01-20:
> May also need to consider other overrides, eg bandwidth & coding since those won't have any specific region.
> 
> @caveman99 I'd advocate for a prescribed frequency / modem configuration. This won't be the last time this comes up.

The modem configuration ranges (both our presets and freely configurable parameters) are the same across all sub ghz Lora chips. so this really needs a custom frequency settable. Radiolib will do a bounds check anyway, i don't see the need to make Coding Rate, Bandwidth and Spreading Factor 'freeform'.

2.4 GHz is ready, but the 'bandwidth' settings is different for the higher bands. Our presets were adjusted accordingly, but the same as for the sub-GHz bands applies.

code8buster, 2023-01-23:
CR/BW/SF already free-form in the way that we already have config.lora protobufs for modem settings if a user doesn't want to use a preset like LongFast.

closed 2023-05-25 by thebentern