meshtastic/firmware #2150: [Feature Request]: Expose connection status (i.e. WiFi, IP, Bluetooth) as part of interface.myInfo

tb0hdan, 2023-01-16:
### Platform

ESP32

### Description

Hi,

Since I'm running bot and sometimes WiFi gets stuck it would be nice to get this information as part of interface.myInfo (as exposed by Python library) so that I can automate device reboots. Right now I use ping to detect this which is not optimal.

```
[5137996][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[5137997][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 203 - ASSOC_FAIL
[5138000][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
[5138013][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a
```

Firmware: `v2.0.13.7e27729-d`

Also, can I get device uptime there as well? Not just reboot count.

garthvh, 2023-01-16:
The my info packet does not go over the mesh and is only received on initial connect.

tb0hdan, 2023-01-16:
Yes, but I still can get it over USB. Is it always the same while device is up and running? There may be some other places to return actual state

caveman99, 2023-01-16:
i think the myNodeInfo is only sent during the initial want_config

caveman99, 2023-01-16:
there was a new buf added returning routing/congestion info. If i read that correctly, it was for IP tunneling over the Mesh. Maybe this can be extended to include that info.

tb0hdan, 2023-01-16:
What I really want is something like 

```JSON
"connectionStatus":{"WiFi":{"IP": "1.2.3.4", "status": "connected", "AP": "dlink"}, "Bluetooth": {"PIN": 123456, "status": "unavailable"}, "Serial": {"speed": 115200, "status": "available"}}
```

thebentern, 2023-01-18:
@tb0hdan, I really like this idea, but I think it deserves it's own bespoke admin message. If you create an issue over in the protobufs repo, I can get started on it.

karamo, 2023-01-22:
> What I really want is something like
> 
> ```json
> "connectionStatus":{"WiFi":{"IP": "1.2.3.4", "status": "connected", "AP": "dlink"}, "Bluetooth": {"PIN": 123456, "status": "unavailable"}, "Serial": {"speed": 115200, "status": "available"}}
> ```

Also WiFi.RSSI please.

tb0hdan, 2023-01-31:
@thebentern Created protobufs issue per request