meshtastic/firmware #2154: [Bug]:

tb0hdan, 2023-01-16:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.0.13.7e27729-d

### Description

T-Beam keeps loosing WiFi connection after flashing 2.0.13.7e27729-d instead of 2.0.12

```
[9185594][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[9185595][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
[9185598][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
[9193335][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[9193336][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
[9193339][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
```

Can be cured temporarily by reboot

tb0hdan, 2023-01-16:
Related to #2150

garthvh, 2023-01-16:
Duplicate of https://github.com/meshtastic/firmware/issues/1946

closed 2023-01-16 by garthvh