meshtastic/firmware #2155: [Feature Request]: Toggle 3v3_S power line via IO2 for RAK devices

tropho23, 2023-01-16:
### Platform

NRF52

### Description

Enable software toggle of 3v3_S power, via IO2 high/low settings to enable/disable power for RAK-based devices.

The intent is to toggle power to the RAK1910 or RAK12500 GPS modules; adding this feature would allow users to get the maximum runtime from installed batteries, when sending real-time GPS updates is not required or desired.

Users would need to be made aware that disabling 3v3_S power cuts off power _**ALL**_ sensors using Slots A-D for RAK19003/19007/5005 and Slots A-F for RAK19001.

caveman99, 2023-01-16:
I actually have a clever idea how we can leverage that. Wire this up to the double-press on user button or switching off through protos like for the AXP192. Switch it on briefly during boot, to auto-detect stuff on the bus. If we find an I2c Sensor (bme680...) briefly switch on the power rail for doing measurements, then back off again. Only do that if the general toggle is in 'off' state

tropho23, 2023-01-16:
> I actually have a clever idea how we can leverage that. Wire this up to the double-press on user button or switching off through protos like for the AXP192. Switch it on briefly during boot, to [...]

Would this be in addition to the global power toggle for sensor slots?  If so this is a great idea indeed, since it would allow the  use to quickly enable power to GPS without having to use the mobile app, which has even greater value for standalone devices without a screen.  For standalone devices there would be no need to add a toggle setting within the on-screen UI, which is very helpful.

caveman99, 2023-01-16:
yes, it would only switch IO2 and therefor all peripherals. But it would behave the same on both platforms, with the added funkctionality to switch power back on briefly for sensor measurements. The few seconds GPS will be active is a tradeoff...

geeksville, 2023-01-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/rak-4631-rak-12500-gps-on-rak19003-energy-hungry/7050/12

garthvh, 2023-01-17:
Turning off the user button in any case seems like a really problematic UX

tropho23, 2023-01-18:
While the user button works, trying PR #2158 causes all sort of trouble: disabled serial module, bootloops, no way to set region due to lack of serial connection...but mostly the bootloops.  I think a hardware switch that shorts IO2 to GND is the best way to go for this, and users who implement that should know what effect it will have on other modules besides GPS.   Thanks to all for considering this and commenting.

I recommend we close this request/issue.

closed 2023-01-30 by thebentern