meshtastic/firmware #2156: [Feature Request]: Rotary select should send the first message if up/down are not configured

bmidgley, 2023-01-16:
### Platform

ESP32

### Description

Turning on canned messages but not defining up/down actions should allow us to have just the one rotary select GPIO send a default (first defined) messages with just one press of the button.

This is especially useful if a sensor needs to trigger a message. In our case, we want the ptt button on a walkie-talkie wired up with the t-beam so it can fire off a message. It's so we can tell who is talking on the analog walkie talkies. The audio is out of band of course.

garthvh, 2023-01-16:
The canned messages module does not engage until the rotary is turned.

Why are you not just using the serial module to send in a message?

bmidgley, 2023-01-16:
@garthvh doesn't the serial module require me to add another microcontroller? What I have is a button press sensor. What I'm trying to do might be a better fit in the telemetry section if we also had a type of sensor that is a simple gpio-connected input.

garthvh, 2023-01-17:
In virtually every case starting with the serial module and sending messages is the way to go for custom functionality, there are a limited number of built in telemetry sensors.  For the built in telemetry they need to be of a specific set of data fields currently and the sensor needs to be instant read, no processing required.

bmidgley, 2023-01-17:
@garthvh thanks, I'm ok with closing this. I could make the added microcontroller small like an attiny85. And as I think about it, in our use case, it might also help to have other kinds of messages besides started-talking like finished-talking or other unrelated stuff.

caveman99, 2023-01-18:
the micro controller is the luxury route for sure, still modifying the Canned Messages to engage with the button press is a viable solution too. While you usually have the send button hooked to the user button, this can be configured differently. So if the user_button is different from the SELECT gpio, the select one could trigger the sending of the first message with 2 pulses.
Right now it's NOT wired up that way, but it could be. Adds a bit of complexity, so if you wanna go the micrcontroller route i'd gladly leave that out.