meshtastic/firmware #2159: [Bug]: Double Click to turn off GPS is disabling GPS too frequently

garthvh, 2023-01-17:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.0.0.0

### Description

Since we added the double click to GPS I am constantly accidentally disabling my GPS, I think we may need to re-think the button part of this feature.

caveman99, 2023-01-17:
the reception when we introduced it was mostly positive, i think the plan to swap the function for 2 and 3 clicks is a good one.

code8buster, 2023-01-19:
The functionality hooks into the position config in memory. If you set position.gps_enabled=0 a Tbeam will shut off its GPS.  It will also turn it back on when set back to 1.

closed 2023-01-20 by thebentern