meshtastic/firmware #2160: [Boarard]: RAK4631 + RAK 19001

ceskip, 2023-01-18:
### SOC

NRF52

### Lora IC

RAK4631

### Product Link

https://store.rakwireless.com/pages/wisblock

### Description

The unit works partially by flashing plain RAK4631 firmware:
BME680 works, but ETHERNET (RAK 13800) and GPS (RAK 1910) doesn't work.
Could it be just an issue of GPIO pins mapping for the RAK 19001 ?

caveman99, 2023-01-18:
RAK 19001 works fine for our firmware, it's not wired differently than the smaller baseboards. There's just careful planning needed for the slot assignments, e.g. the ESP32 core (11200), the external LoRa Module and Ethernet won't work casue there's pin collisions. Without looking, there may be collisions with the GPS module and Ethernet. My working model here does not have GPS onboard. But this would also be true for a RAK19007

closed 2023-01-18 by caveman99

ceskip, 2023-01-18:
Can you help me to understand better the pin assignment on 19001 ?