meshtastic/firmware #2165: [Bug]: Heap-Leak

karamo, 2023-01-18:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.0.x

### Description

All FW 2.0.x versions contain several heap leaks.

After several days of in-depth troubleshooting/analysis, my realization was that many error symptoms that occur after a certain period of operation are likely due to multiple heap leaks.
They occur with various symptoms when the heap is no longer sufficient.
Many reboots run unnoticed and are not noticed during operation, except that individual messages may be lost. Sometimes, however, malfunctions occur, e.g. reboot loop

My solution/symptom fix is now such that the heap is monitored (watchdog). If the heap falls below a certain value, a REBOOT is triggered = "Garbage collection"

1 T-Beam and 2 TLORA 2.1_1.6 have been running fine with this modification since days, i.e. WiFi & MQTT now work perfectly.

karamo, 2023-01-18:
[image]
Here you kann see the planned "Heap-Reboot" of my T-Beam and TLORA.

caveman99, 2023-01-20:
@karamo care to send a Pull request with your changes for review?

karamo, 2023-01-20:
I am not able to create a PR. I can only specify the program sequence I used that works. Please implement it according to the FW.
Delay and WiFi.disconnect should also be installed. Here ist the whole
 **void OSThread::run()**

`void OSThread::run()
{
#ifdef DEBUG_HEAP
    auto heap = ESP.getFreeHeap();
#endif    
    currentThread = this;
    auto newDelay = runOnce();
#ifdef DEBUG_HEAP
    auto newHeap = ESP.getFreeHeap();
    if (newHeap < heap)
        LOG_DEBUG("------ Thread %s leaked heap %d -> %d (%d) ------\n", ThreadName.c_str(), heap, newHeap, newHeap - heap);
    if (heap < newHeap)
        LOG_DEBUG("++++++ Thread %s freed heap %d -> %d (%d) ++++++\n", ThreadName.c_str(), heap, newHeap, newHeap - heap);

    if (newHeap < 10000) {
        // Get our hardware ID
        uint8_t dmac[6];
        getMacAddr(dmac);
        char mac [18];
        sprintf(mac, "!%02x%02x%02x%02x", dmac[2], dmac[3], dmac[4], dmac[5]);
        std::string heapTopic = "msh/2/heap/"+std::string(mac);
        std::string heapString = "0";
        mqtt->pubSub.publish(heapTopic.c_str(), heapString.c_str(), false);
        // delay
        //LOG_DEBUG("\n\n====== WiFi.disconnect ... ======\n");
        //WiFi.disconnect();
        LOG_DEBUG("\n\n====== heap too low [10000] -> reboot in 5s ======\n\n");
        screen->startRebootScreen();
        rebootAtMsec = millis() + 5000;
    }
#endif

    runned();

    if (newDelay >= 0)
        setInterval(newDelay);

    currentThread = NULL;
}
`

karamo, 2023-01-20:
I use this for monitoring the "Heap-Reboot" on my MQTT-Server:

`// Get our hardware ID
    uint8_t dmac[6];
    getMacAddr(dmac);
    char mac [18];
    sprintf(mac, "!%02x%02x%02x%02x", dmac[2], dmac[3], dmac[4], dmac[5]);
    std::string heapTopic = "msh/2/heap/"+std::string(mac);
    std::string heapString = "0";
    mqtt->pubSub.publish(heapTopic.c_str(), heapString.c_str(), false);`

PR #2191 (2023-01-23): tempfix: reboot ESP Node if heap runs low.
fixes #2165
commit 3b804214 (2023-01-23): tempfix: reboot ESP Node if heap runs low.
fixes #2165
commit 8e0dcb09 (2023-01-24): tempfix: reboot ESP Node if heap runs low.
fixes #2165
closed 2023-01-23 by caveman99