meshtastic/firmware #2167: [Bug]: Crash when sending serial data in text mode

bmidgley, 2023-01-18:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

2.0.13.7e27729-d

### Description

Sending serial data causes an immediate reboot.

The serial module is trying to send everything as text messages.

I'm sending from an esp8266 nodemcu running micropython if that helps. Sending from its hardware uart. I was able to separately verify the stream coming through at 115k from a pi direct to its uart. The micropython is in https://gist.github.com/bmidgley/bf7a1c4a96e55029d3c0e51c1dc0757f

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 34, "txGpio": 12 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "US", "hopLimit": 3, "txEnabled": true, "txPower": 30 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": { "enabled": true, "rxd": 14, "txd": 25, "baud": "BAUD_115200", "timeout": 1, "mode": "TEXTMSG" }, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900, "environmentMeasurementEnabled": true, "environmentScreenEnabled": true }, "cannedMessage": { "allowInputSource": "_any" } }

### Relevant log output

```Shell
(base) bmidgley@MacBook-Air-2 esptool % meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:511 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:389 Not sending packet because protocol use is disabled by noProto
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).

Core  1 register dump:
PC      : 0x40081b97  PS      : 0x00050035  A0      : 0x40086f10  A1      : 0x3ffc010c
A2      : 0x00004000  A3      : 0x3ffbe718  A4      : 0x00000001  A5      : 0x40097982
A6      : 0x3ffda170  A7      : 0x3ffda1bc  A8      : 0x00000001  A9      : 0x3ffc00ec
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000004
A14     : 0x3ffd5a74  A15     : 0x80000001  SAR     : 0x00000012  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core  1 was running in ISR context:
EPC1    : 0x4014730f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x40081b94:0x3ffc010c |<-CORRUPTED

Core  0 register dump:
PC      : 0x4021b9c6  PS      : 0x00060835  A0      : 0x801450f8  A1      : 0x3ffbda40
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x8009b574  A5      : 0x3ffbd960
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80144c3a  A9      : 0x3ffbda10
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009b574  A13     : 0x3ffdfcc0
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4021b9c3:0x3ffbda40 0x401450f5:0x3ffbda60 0x40098eec:0x3ffbda80

ELF file SHA256: 183cbd88d33fad19

E (36842) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
E (810) esp_core_dump_flash: No core dump p?\W?????found!
E (810) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   463][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
WARNING file:mesh_interface.py _sendToRadio line:511 Not sending packet because protocol use is disabled by noProto
INFO  | ??:??:?? 0

//\ E S H T /\ S T / C
```

garthvh, 2023-01-18:
You want the simple mode probably, the txtmsg mode is designed to be an easy way to send a text message, so it does indeed send everything as a text message. There is also a simple serial mode, NMEA and protos modes.

bmidgley, 2023-01-18:
@garthvh are there more docs for simple mode? I can't tell what it's expecting.

It looks like I get three empty messages in mqtt with my existing code after it sends the two strings.

```
mosquitto_sub ... -F %J
{"tst":"2023-01-18T19:07:31.509855Z+0000","topic":"msh/2/json/LongFast/!55c7311c","qos":0,"retain":0,"payloadlen":108,"payload":{"channel":0,"from":-1543808584,"id":65902487,"sender":"!55c7311c","timestamp":1674068851,"to":-1,"type":""}}
{"tst":"2023-01-18T19:07:39.716064Z+0000","topic":"msh/2/json/LongFast/!55c7311c","qos":0,"retain":0,"payloadlen":108,"payload":{"channel":0,"from":-1543808584,"id":65902489,"sender":"!55c7311c","timestamp":1674068859,"to":-1,"type":""}}
{"tst":"2023-01-18T19:07:47.200294Z+0000","topic":"msh/2/json/LongFast/!55c7311c","qos":0,"retain":0,"payloadlen":108,"payload":{"channel":0,"from":-1543808584,"id":65902490,"sender":"!55c7311c","timestamp":1674068866,"to":-1,"type":""}}

bmidgley, 2023-01-18:
At least one similar error happened because the interrupt handler was taking too long. Maybe there is network communication in the interrupt handler? I'm not sure where the code is though.

https://stackoverflow.com/questions/71992044/esp32-core-1-paniced-interrupt-wdt-timeout-on-cpu1

```
... serial communication within an interrupt handler ... does not work.

When an interrupt handler is called, it blocks the entire system. So it must only do minimal work and return quickly. Serial communication however is slow and causes long delays. Thus, your program is terminated by the watchdog ("wdt" is the abbreviation for the watchdog).

bmidgley, 2023-01-19:
It doesn't crash if the messages are short--each 2 characters, "xx" and "oo". It doesn't crash, but messages sent close together get into the same text. Is there any way the newline could kick off the message to shorten the latency?

I think in the crash above, the two long messages are sent together and somehow transmitting that longer thing takes so long that the watchdog spots it and kills it.

```
{"tst":"2023-01-19T23:27:57.433345Z+0000","topic":"msh/2/json/LongFast/!55c7311c","qos":0,"retain":0,"payloadlen":177,"payload":{"channel":0,"from":-1543808584,"id":507859756,"payload":{"text":"xx\r\noo\r\n"},"sender":"!55c7311c","timestamp":1674170876,"to":-1,"type":"text"}}

caveman99, 2023-01-20:
there is no serial activity in an ISR. There was a change to allow sending binary chunks over serial. I think this inhibits the old behaviour where after the cr the string was sent. That would explain your beahviour. I think we need to look at this again and define a wanted behaviour.

garthvh, 2023-08-09:
Is this still happening @bmidgley there are some updates around the serial modes in some recent releases.

closed 2023-09-13 by thebentern