meshtastic/firmware #2171: [Bug]: Probably bug in calculation of channel load

D4rk4, 2023-01-18:
### Category

Other

### Hardware

T-Lora v1

### Firmware Version

2.0.13

### Description

Preferences: { "device": { "role": "ROUTER_CLIENT", "serialEnabled": true }, "position": { "positionBroadcastSecs": 120, "positionBroadcastSmartEnabled": true, "fixedPosition": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 3600, "positionFlags": 3, "rxGpio": 34, "txGpio": 12 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 4294967295, "sdsSecs": 10, "lsSecs": 300, "minWakeSecs": 31104000 }, "network": { "wifiEnabled": true, "wifiSsid": "DarkSkyI", "wifiPsk": "kolokol9kolokol", "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_868", "hopLimit": 5, "txEnabled": true, "txPower": 27, "overrideDutyCycle": true }, "bluetooth": { "fixedPin": 123456 } }

### Relevant log output

```Shell
WARN  | 21:26:48 3077 [DeviceTelemetryModule] Tx air utilization is >0 percent. Skipping this opportunity to send.
```

GUVWAF, 2023-01-19:
The problem is only with the way how it is printed. Should be fixed with #2175.

D4rk4, 2023-01-19:
@GUVWAF may you check, does it's realy a cosmetic issue? I start investigating it because my node stopt tx.

GUVWAF, 2023-01-19:
At his point it will indeed not send this DeviceTelemetry packet, because it exceeded 50% of the duty cycle. Other packets, such as text or position will still be sent.

PR #2175 (2023-01-20): Fix two print format specifiers
Fixes #2171.
closed 2023-01-20 by thebentern