meshtastic/firmware #2181: [Bug]: Remote Node Administration without permission

nukevoid, 2023-01-20:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.0.9

### Description

An altered packet with "from == 0" can do Remote Node Administration without permission.
Any node with hacked firmware can build a custom packet with 'from = 0' and portnum = ADMIN_APP_VALUE and control all nodes in the network. It works without creating any admin channel. 
Tested on T-Beam v1.1

nukevoid, 2023-01-20:
The problem is in also in line `bool rxChannelOk = !pi.boundChannel || (mp.from == 0) || (strcasecmp(ch->settings.name, pi.boundChannel) == 0);` in MeshModule::callPlugins. rxChannelOk  will be 'true' is case 'mp.from == 0'

GUVWAF, 2023-01-20:
Good find! 

According to the comments, checking for `mp.from == 0` was put in intentionally there to allow apps to configure the node without having an admin channel. But indeed, if you alter a LoRa packet with `from = 0`, you can do administration without permission. So ignoring a packet deep in the RadioLibInterface as you propose in your PR #2179 seems fine to me. Just did a quick check that it doesn't break phone connections.

garthvh, 2023-01-20:
Using the node number and not 0 will probably require client changes as you get a warning in the serial log if you send a node number for local admin messages so clients often use a 0 to suppress the warning.

caveman99, 2023-01-20:
@garth that should still work, since the filter in #2179 is really early in the receive path of radiolib / lora. Packets over PhoneAPI don't pass through there.

GUVWAF, 2023-01-21:
Fixed by #2179.

closed 2023-01-21 by GUVWAF