meshtastic/firmware #2193: [Bug]: Heltec lora32 v2.1 with neo6m gps module  "NO GPS"

LO77ARO, 2023-01-23:
### Category

Other

### Hardware

Heltec v2.1

### Firmware Version

2.0.10

### Description

i had 4 heltec lora32 v2.1, two with version 1.2.65. and two with 2.0.10.  I installed neo6m gps modules in the two with 1.2.65. Using rx to pin 36 and tx to pin 37. It worked perfect! In the other two modules, with 2.0.10, I use the same pins 36 and 37 and it doesn't work. 
I looked at the variant.h(v 2.0.10) file and found that the pins were changed to 36 rx and 33 tx. So I re-soldered the correct pins but it still doesn't work.

Then I updated the modules where the gps worked to version 2.0.10 and the modules continued to work even when connected to pins 36 and 37.

i cant understand what im doing wrong.

caveman99, 2023-01-23:
only pin 36 is used for those really. 37 was a wrong allocation a this pin is input only, so it was changed to 33.

You may need to factory reset the chips for them to start working in 2.0.x since we changed the protocol from UBX to NMEA. Connect the chips to the node with the right pins (36/33), Power the node on, wait for 2 mintes, power off, power on, 2 minutes, power off, power on. they should work after that procedure.

closed 2023-01-23 by caveman99

LO77ARO, 2023-01-24:
Thanks!  i fixed that way!