meshtastic/firmware #2197: [Feature Request]: Coords force transfer

Andrey-VD, 2023-01-24:
### Platform

ESP32

### Description

I suggest adding an option to send coordinates when transmitting a message. This can be very important for moving objects (e.g. cars)
You can implement it either by adding a coordinate field in the message, or by forcibly initiating the NodeInfo transfer.

garthvh, 2023-01-24:
Position is a separate packet that can be managed for vehicles already. What are you unable to do with the existing position config?  Adding position data to messages would increase channel utilization.

Andrey-VD, 2023-01-24:
> Position is a separate packet that can be managed for vehicles already. What are you unable to do with the existing position config? Adding position data to messages would increase channel utilization.

If I understood correctly, the coordinates are sent only by NodeInfo packets with low priority. If you want to understand from which location the car sent the message, you need to configure the frequent sending of coordinates. It is debatable whether the communication channel will take up more - frequent sending of NodeInfo packets or more rare sending of messages including coordinates

garthvh, 2023-01-24:
There is a position packet that is not a part of the node info that is what you are interested in.

Andrey-VD, 2023-01-24:
> There is a position packet that is not a part of the node info that is what you are interested in.

Of course, you're right about the packet with coordinates. However, I continue to believe that rare packets with messages with connected coordinates will occupy the channel less than the forced frequent sending of coordinates. Maybe you should make the option "send messages with coordinates"?

garthvh, 2023-01-24:
Smart position handles this already.  I think the node info with position you are referencing is only sent over the phone api and not the mesh.

closed 2023-01-25 by garthvh