meshtastic/firmware #2200: [Bug]: Cannot set power.ls_secs - it is always 300 secs (client) or 86400 (router)

rpsainio, 2023-01-24:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.13.7e27729-d

### Description

the $SUBJECT tells it all: it is not possible to set that value with CLI.

### Relevant log output

```Shell
meshtastic --set power.lsSecs 200 --port=/dev/ttyACM0
Connected to radio
Set power.ls_secs to 200
Writing modified preferences to device
[risto@rsa46 configs]$ meshtastic --get power.lsSecs --port=/dev/ttyACM0 
Connected to radio

power.ls_secs: 86400
Completed getting preferences
[risto@rsa46 configs]$ meshtastic --set device.role  CLIENT_MUTE
Connected to radio
Set device.role to CLIENT_MUTE
Writing modified preferences to device
[risto@rsa46 configs]$ meshtastic --get power.lsSecs --port=/dev/ttyACM0 
Connected to radio

power.ls_secs: 300
Completed getting preferences
[risto@rsa46 configs]$ meshtastic --version
2.0.9
```

garthvh, 2023-01-24:
Power settings should be managed by the role now

caveman99, 2023-01-24:
the explicit setting should still be honored, even if in most cases it does more harm than good.

rpsainio, 2023-01-24:
explains as I checked the source already - those default values look quite arbitrary to me.

rpsainio, 2023-01-24:
maybe we need : "this parameter will be become obsolete in near future"

rpsainio, 2023-01-24:
Ah whatever, the best SW is the one you bake yourself.. I noticed this when I tried to test further #2084 as that boot-loops happens only when node goes to LS and maybe there is time dependence. @caveman99 cheers from HD

closed 2023-01-24 by rpsainio

garthvh, 2023-01-25:
> 

Do no harm!