meshtastic/firmware #2202: [Feature Request]: add configurable setting for gps position reporting

djscoutmaster, 2023-01-25:
### Platform

ESP32

### Description

Make the hard coded 40% airtime gps message cutoff configurable for location specific use cases.

caveman99, 2023-01-25:
There's discussion at the moment to create a new device role 'tracker' that would incorporate this and other tweaks.

Andrey-VD, 2023-01-26:
> 

Everyone has different needs. For example, it is the tracker functions that are most important to me. At the same time, voice transmission functions are absolutely not needed, although codecs are already appearing in development.

GUVWAF, 2023-01-29:
The 'Tracker' role is now in the 2.0.15 alpha release. From Discord: 
> The Tracker role increases the priority of Position packets sent out over the mesh as well as setting the position broadcast interval to 2 minutes with a 1 minute gps update interval

You can try it using the CLI with `--set device.role 5 `.

thebentern, 2023-01-29:
I believe that Increasing the packet priority via the new Tracker role will have a better chance for successful delivery of position than removing the 40% airtime limit, which will only lead to more packet collisions. Another option we might consider looking at later is providing a sub mode of tracker where position messages don't want acks (ala fire and forget) to decrease the overall airtime footprint.

closed 2023-01-29 by thebentern

mc-hamster, 2023-01-29:
@thebentern Brilliant! That's a good bit of intel. There are use cases where the network overhead may not be properly tuned.

@Andrey-VD As an open source project, we build on a community of people who contribute their time and passion to the project. If there is something that doesn't interest you, don't use it but please be supportive of our contributors.