meshtastic/firmware #2203: [Bug]:  Router without power saving mode

SR7673, 2023-01-25:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.13

### Description

Not really a heavy bug, but:
Referring to https://meshtastic.discourse.group/t/router-and-client-what-is-the-difference/7060/5

In PowerFSM.cpp, line 23, it looks like isPowerSavingMode is always true to the node is running as a router. And thus, isPowered (line 12) is always false in router mode.

I expected that, if power saving is disabled, then it IS disabled. However in router mode it seems to be enabled all the time athough the CLi and the app shows that it is disabled.

Has has been annoying recently to see that this node always feel into deep sleep mode. For now i will put it into CLIENT mode however its is confusing that node saves power although this has been disabled.

If there is a lack of power (small solar module or so), people can enable power saving mode, if they have enough power, they can disable...

geeksville, 2023-01-25:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/router-and-client-what-is-the-difference/7060/6

garthvh, 2023-01-25:
This is what router client is for

charminULTRA, 2023-01-25:
Power saving mode turns off the BLE/Wifi radios and the screen. When using as a Router, the intent is that it's just a standalone device that a person won't be controlling directly, so the behavior and code are correct, just difficult to interpret.

closed 2023-01-26 by garthvh

krausedj, 2024-07-17:
With router client removed/deprecated should this be reconsidered?