meshtastic/firmware #2206: [Bug]: "heap too low [10000] -> reboot in 5s" does not reboot in given time

rpsainio, 2023-01-25:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.14.cb8532f2-d

### Description

There is a new feature designed to reboot the node when heap gets below 10000 #2165 . Unfortunately this does not work as designed and node did not reboot. Shortening the time it waits before rebooting seems to solve the problem

[reboot_01.txt](https://github.com/meshtastic/firmware/files/10504142/reboot_01.txt)

[reboot_02.txt](https://github.com/meshtastic/firmware/files/10504143/reboot_02.txt)

karamo, 2023-01-25:
Yes, I've also noticed that it's awkwardly placed in **OSThread.cpp**. If this function is called within the 5s, then it comes to a retrigger and delays the reboot.
Why did this work for me?
Probably because I compiled myself with **DEBUG_HEAP** and additionally had the **DEBUG output turned on**.
My suggestion would now be to move this heap test to the **power .cpp**, since it is only called every 30s. Also, both the value 10000 is only estimated by me, since 7000 was too low.
Also, the 5s are just any value and can be reduced arbitrarily.

rpsainio, 2023-01-26:
@karamo I just tested asserting and it seems to work as well, meaning no boot-loops etc. Once low on heap I would not waste time waiting that somebody releases heap rather than rebooting. Will cost some extra messages sent around. According to my test this  happens after ca. 16000 seconds == 4 hours so even an MS-windows survives longer..

karamo, 2023-01-27:
I can confirm it now.
As the last test run on the TLORA 2.1_1.6 board has shown, 10000 heap reserve for reboot is not enough and 5s delay to reboot is too long. It crashed ...
My recommendation:
1) Move the query to Power.cpp
2) set it to 11000
3) set the delay to max. 1s

karamo, 2023-01-27:
Sorry, correction, the **30s** rate at which **[Power] thread** runs can also be too long.
If the **[Blink] thread** runs at 1s, the query would be much better placed here.

caveman99, 2023-01-27:
see #2210

caveman99, 2023-01-30:
fixed in 2.0.14 beta

closed 2023-01-30 by caveman99