meshtastic/firmware #2208: [Bug]: [RadioIf] SX126x standby failed with error -707

Andrey-VD, 2023-01-26:
### Category

Hardware Compatibility

### Hardware

DIY

### Firmware Version

2.0.12.2400dd4-d

### Description

reboot after almost every message sent or received. 
Module E22400M30S

--------------------------------------------------------------
Connected to radio
iceTelemetryModule] txGood=0,rxGood=0,rxBad=0

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Using channel 0 (hash 0x8)

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Expanding short PSK #1

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Using AES128 key!

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] ESP32 crypt fr=84de2c0c, num=7416aa1f, numBytes=11!

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] decoded message (id=0x7416aa1f fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)

**### DEBUG | ??:??:?? 12 [RadioIf] SX126x standby failed with error -707**

assert failed: void SX126xInterface<T>::setStandby() [with T = SX1268] SX126xInterface.cpp:175 (err == RADIOLIB_ERR_NONE)

Backtrace: 0x4008448d:0x3ffd8540 0x400966a1:0x3ffd8560 0x4009c821:0x3ffd8580 0x400e053b:0x3ffd86b0 0x400dfaf8:0x3ffd86d0 0x400e4335:0x3ffd86f0 0x400e4394:0x3ffd8710 0x4020fd42:0x3ffd8730 0x400d5685:0x3ffd8750 0x400d573a:0x3ffd8770 0x400fb2fd:0x3ffd8790 0x400de3cd:0x3ffd87c0 0x4012f329:0x3ffd87e0

ELF file SHA256: a3d62f1cb6fab2f1

E (18444) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

### Relevant log output

```Shell
Connected to radio
iceTelemetryModule] txGood=0,rxGood=0,rxBad=0

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Using channel 0 (hash 0x8)

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Expanding short PSK #1

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] Using AES128 key!

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] ESP32 crypt fr=84de2c0c, num=7416aa1f, numBytes=11!

DEBUG | ??:??:?? 12 [DeviceTelemetryModule] decoded message (id=0x7416aa1f fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)

DEBUG | ??:??:?? 12 [RadioIf] SX126x standby failed with error -707

assert failed: void SX126xInterface<T>::setStandby() [with T = SX1268] SX126xInterface.cpp:175 (err == RADIOLIB_ERR_NONE)

Backtrace: 0x4008448d:0x3ffd8540 0x400966a1:0x3ffd8560 0x4009c821:0x3ffd8580 0x400e053b:0x3ffd86b0 0x400dfaf8:0x3ffd86d0 0x400e4335:0x3ffd86f0 0x400e4394:0x3ffd8710 0x4020fd42:0x3ffd8730 0x400d5685:0x3ffd8750 0x400d573a:0x3ffd8770 0x400fb2fd:0x3ffd8790 0x400de3cd:0x3ffd87c0 0x4012f329:0x3ffd87e0

ELF file SHA256: a3d62f1cb6fab2f1

E (18444) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

Rebooting...

ets Jul 29 2019 12:21:46
[... 930 lines cut ...]

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13104

load:0x40080400,len:3036

entry 0x400805e4

E (791) esp_core_dump_flash: No core dump ?f????????found!

E (791) esp_core_dump_flash: No core dump partition found!

[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

INFO  | ??:??:?? 0
```

caveman99, 2023-04-02:
this will be fixed in release 2.1.7

closed 2023-04-02 by caveman99

SCWhite, 2024-04-12:
@caveman99  
HI, Can you kindly explain how is this being fixed?  

I'm building a diy hardware with :  
Adafruit nRF52840 Feather + sx1262  Lora chip  

Testing with Firmware Version:  
2.3.4.ea61808  

Description:  
device will boot, but when I try to change reign setting with Meshtastic APP  
it'll crash and reboot  

Showing  
```
DEBUG | ??:??:?? 21 [Router] SX126x standby failed with error -707
ERROR | ??:??:?? 21 [Router] assert failed src\mesh\SX126xInterface.cpp: 222, void SX126xInterface<T>::setStandby() [with T = SX1262], t
```

Firmware before 2.1.11 works fine, I'm guessing has something to do with new parameter in  
```RadioLibInterface(hal, cs, irq, rst, busy)```

Any thing helps, thanks!

GUVWAF, 2024-04-12:
@SCWhite 

See:
https://github.com/jgromes/RadioLib/blob/d77823375a3c4329a1a1182fcf4787f008878810/src/modules/SX126x/SX1262.h#L42

Do you have `#define SX126X_DIO3_TCXO_VOLTAGE` set in your variant file? Depending on whether you module has a TCXO or not, you should set this to the correct voltage.

SCWhite, 2024-04-12:
@GUVWAF 

Thanks for the fast reply, I've tried delete that define / set to 0 but I still get the same error  
I'll try to ground DIO3, make sure TCXO isn't working.

Thanks, will report back soon!

SCWhite, 2024-04-12:
HI @GUVWAF 

So, this is what I find out.  

According to the datasheet:   
"set pin to HIGH to enable TCXO, LOW to put TCXO to sleep"  

[image]

I followed most design by not connecting DIO3, hence ```#define SX126X_DIO3_TCXO_VOLTAGE``` should be 0V
and to make sure I'm not using it, I even try to solder that pin directly to GND  

but I still get  
```
SX126x standby failed with error -707
```

it turns out, the TCXO was actually connected in the chip, internally.   
so I had to do  ```#define SX126X_DIO3_TCXO_VOLTAGE 1.8``` to make it work.

never had thought of that.

Thanks for the advice, you saved me tons of time and brain cell.
You're the best !