meshtastic/firmware #2227: [Bug]: Possible lower heap leak if lora.tx_enabled false

karamo, 2023-01-30:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.13

### Description

First, I tested whether **WiFi** or **WiFi+MQTT** have an effect on the heap leak.
The heap derating values in all tested cases are 465±2 bytes/minute.
The 3 modules for the test:
blue WZ1 T-Beam SX1276
ret TEST T-Beam SX1262
magenta ZELX TLORA v2.1_1.6 as Referenzmodul

Here in the standard mode Send & Receive enabled.

[image]

One way to restrict the transmission via a parameter is
`--set lora.tx_enabled false`

[image]

**There is a significantly higher heap derating if tx_enabled is set to false.**

caveman99, 2023-01-30:
looks like this was related to https://github.com/meshtastic/firmware/pull/2231 - can you retest after the next firmware release?

karamo, 2023-01-30:
Yes, I will repeat the test with 2.0.15 soon.

karamo, 2023-01-31:
Now I tested on both modules FW 2.0.14 with `meshtastic --port com8 --set lora.tx_enabled false` on the red Module.

[image]

The larger heap leak is fixed. There is even an improvement, but it is also logical, since with lora.tx_enabled false now no transmissions take place at all.

karamo, 2023-01-31:
Is fixed in **FW 2.0.14**
can be closed.

closed 2023-01-31 by thebentern