meshtastic/firmware #2235: [Bug]: 2.0.16 not working...

SR7673, 2023-01-31:
### Category

Hardware Compatibility

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.16

### Description

I just uploaded V2.0.16 to a T-Beam V1.1 and a TTGO Lora V2.1-1.6.
After the upload, the modules do not work at all. No connection in the CLI nor does the module show something in the display. Also, no BT Pin is offered...
(It's working on the RAK4631 though...)

caveman99, 2023-01-31:
how did you flash the modules? also upgrade or wipe?

SR7673, 2023-01-31:
As usual, i use the python CLI on Linux... Worked on the older versions like 2.0.13... 
However i took the "old" device-install.sh. Could this be the reason? I should try...

SR7673, 2023-01-31:
Confirmed: Same result when using the latest device-install.sh delivered in V2.0.16

caveman99, 2023-01-31:
device-install wipes everything. do you have unusual output?

SR7673, 2023-01-31:
No, there is no relevant difference in the output (between 2.0.13 and 2.0.16)

thebentern, 2023-01-31:
I just flashed a T-beam with the device install script and a TLoRA v2 1.6 with the device update script. I'm not seeing any issues

thebentern, 2023-01-31:
If you post the full command and output of the command, we might be able to help debug what's going on.

SR7673, 2023-01-31:
Problem solved. It was my mistake. I took my own build with a minor change. Obviously i made a mistake. Now i even tested and successfully run 2.0.17!!!!!
I should have tried the original version before opening an issue. Sorry.

closed 2023-01-31 by SR7673