meshtastic/firmware #2238: No Heap-Leak with Role REPEATER

karamo, 2023-02-01:
With FW 2.0.14 ther is a new role: REPEATER
Many threads that are not required for the role REPEATER do not start.
As a result, it turns out that basically no heap leak occurs anymore.
The blue one is REPEATER
	meshtastic --port com16 --set device.role REPEATER

[image]

caveman99, 2023-02-01:
related to #2228

thebentern, 2023-02-01:
Closing this, since it's more of an observation and so we can track the heap leak issue in #2228

closed 2023-02-01 by thebentern

karamo, 2023-02-01:
The difference between **CLIENT** and **REPEATER** are the following threads, which do not run with REPEATER:
```
• GPS
• NodeInfoModule
• PositionModule
• DeviceTelemetryModule
```
In subsequent test runs, the effects are tested individually.