meshtastic/firmware #2239: [Bug]: Heap-Leak in DeviceTelemetryModule

karamo, 2023-02-01:
### Category

Other

### Hardware

T-Beam, Other

### Firmware Version

2.0.17

### Description

As shown in issue #2238, no heap leaks occur with role REPEATER. In subsequent test runs, the effects are tested individually. The difference between CLIENT and REPEATER are the following threads, which do not run at role REPEATER:
• GPS
• NodeInfoModule
• PositionModule
• DeviceTelemetryModule

[image]
GPS, NodeInfoModule, PositionModule = no Heap-Leak

[image]
(red) TEST = CLIENT
(magenta) ZELX = CLIENT (magenta) TLORA (Referenz)
(blue) WZ1 = REPEATER + DeviceTelemetryModule

karamo, 2023-02-02:
I found the heap-leak. @caveman99 will make the PR.
[image]

mc-hamster, 2023-02-02:
WOW, WOW, WOW!!!!!

Thank you for this great research!

commit a13adfb5 (2023-02-02): fixes #2239
closed 2023-02-02 by caveman99