meshtastic/firmware #2242: [Bug]: RAK 4631 with ethernet crashes on MQTT msg

Kollensperger, 2023-02-02:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.14 and 2.0.17

### Description

Rak 4631 (both on 5005 and 19007 baseboards) with ethernet and MQTT crashes after a short while. This is a known good working config on a t-beam, but instead of wifi enabled, ethernet is enabled in the RAK 4631.
Impossible to send more than 1 MQTT message to the board.
Interestingly enough, even though the message (see log output below) "Testing" was clearly received by the ethernet enabled RAK, the device it was sent from shows both this message and any subsequent messages as "Max retransmission Reached" in iOS.

on occasion the final line in the terminal output is 

`WARNING file:stream_interface.py __reader line:179 Meshtastic serial port disconnected, disconnecting... read failed: [Errno 6] Device not configured`

but this is not always the case. 
This is reproducible on both 2.0.14.2baaad8 and 2.0.17.5d1c06b.
On a separate note it seems the ethernet cable cannot be particularly long - 30 cm long cable yields lit amber and green control lights on the RAK, 1 m long cable yields error message `ERROR | ??:??:?? 67 Ethernet cable is not connected`.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 2 Filesystem files:
WARNING file:mesh_interface.py _sendPacket line:389 Not sending packet because protocol use is disabled by noProto
DEBUG | ??:??:?? 2  channels.proto (252 Bytes)
DEBUG | ??:??:?? 2  config.proto (75 Bytes)
DEBUG | ??:??:?? 2  module.proto (63 Bytes)
DEBUG | ??:??:?? 2  db.proto (316 Bytes)
DEBUG | ??:??:?? 2 Using analog input 5 for battery level
INFO  | ??:??:?? 2 No I2C devices found
INFO  | ??:??:?? 2 Meshtastic hwvendor=9, swver=2.0.14.2baaad8-d
DEBUG | ??:??:?? 2 Reset reason: 0x0
DEBUG | ??:??:?? 2 Setting random seed 2993911210
INFO  | ??:??:?? 2 Initializing NodeDB
INFO  | ??:??:?? 2 Loading /prefs/db.proto
INFO  | ??:??:?? 2 Loaded saved devicestate version 20
INFO  | ??:??:?? 2 Loading /prefs/config.proto
INFO  | ??:??:?? 2 Loaded saved config version 20
INFO  | ??:??:?? 2 Loading /prefs/module.proto
INFO  | ??:??:?? 2 Loaded saved moduleConfig version 20
INFO  | ??:??:?? 2 Loading /prefs/channels.proto
INFO  | ??:??:?? 2 Loaded saved channelFile version 20
INFO  | ??:??:?? 2 No /oem/oem.proto preferences found
INFO  | ??:??:?? 2 Wanted region 3, using EU_868
DEBUG | ??:??:?? 2 region=3, NODENUM=0xf8120fdb, dbsize=3
DEBUG | ??:??:?? 2 Using MSL altitude model
DEBUG | ??:??:?? 2 WANT GPS=1
WARN  | ??:??:?? 3 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 3 GxGSA NOT available
INFO  | ??:??:?? 3 External Notification Module Disabled
WARN  | ??:??:?? 3 SX126xInterface(cs=42, irq=47, rst=38, busy=46)
INFO  | ??:??:?? 3 Starting meshradio init...
INFO  | ??:??:?? 3 Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 3 Set radio: region=EU_868, name=Lights, config=0, ch=0, power=27
INFO  | ??:??:?? 3 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
INFO  | ??:??:?? 3 Radio myRegion->numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 3 Radio channel_num: 0
INFO  | ??:??:?? 3 Radio frequency: 869.525024
INFO  | ??:??:?? 3 Slot time: 42 msec
INFO  | ??:??:?? 3 Set radio: final power level=22
INFO  | ??:??:?? 3 SX126x init result 0
[... 64 lines cut ...]
DEBUG | ??:??:?? 25 [Router] decoded message (id=0x22f15c0e fr=0x53 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1 rxSNR=6.5 rxRSSI=-22)
DEBUG | ??:??:?? 25 [Router] handleReceived(REMOTE) (id=0x22f15c0e fr=0x53 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1 rxSNR=6.5 rxRSSI=-22)
DEBUG | ??:??:?? 25 [Router] Module 'text' wantsPacket=1
INFO  | ??:??:?? 25 [Router] Received text msg from=0xafd2da53, id=0x22f15c0e, msg=Testing 
DEBUG | ??:??:?? 25 [Router] Module 'text' considered
DEBUG | ??:??:?? 25 [Router] Module 'canned' wantsPacket=1
DEBUG | ??:??:?? 25 [Router] Module 'canned' considered
DEBUG | ??:??:?? 25 [Router] Module 'ExternalNotificationModule' wantsPacket=1
INFO  | ??:??:?? 25 [Router] External Notification Module Disabled
DEBUG | ??:??:?? 25 [Router] Module 'ExternalNotificationModule' considered
DEBUG | ??:??:?? 25 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 25 [Router] Received routing from=0xafd2da53, id=0x22f15c0e, portnum=1, payloadlen=8
DEBUG | ??:??:?? 25 [Router] Routing sniffing (id=0x22f15c0e fr=0x53 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1 rxSNR=6.5 rxRSSI=-22)
INFO  | ??:??:?? 25 [Router] Rebroadcasting received floodmsg to neighborsShould encrypt MQTT?: 0
DEBUG | ??:??:?? 25 [Router] publish msh/2/c/Home/!f8120fdb, 71 bytes
DEBUG | ??:??:?? 25 [Router] got text message of size 8
INFO  | ??:??:?? 25 [Router] text message payload is of type plaintext
INFO  | ??:??:?? 25 [Router] serialized json message: {"channel":0,"from":-1345136045,"id":586243086,"payload":{"text":"Testing "},"sender":"!f8120fdb","timestamp":0,"to":-1,"type":"text"
INFO  | ??:??:?? 25 [Router] JSON publish message to `W , 134 bytes: {"channel":0,"from":-1345136045,"id":586243086,"payload":{"text":"Testing "},"sender":"!f8120fdb","timestamp":0,"to"
DEBUG | ??:??:?? 25 [Router] Original length - 8 
DEBUG | ??:??:?? 25 [Router] Compressed length - 5 
DEBUG | ??:??:?? 25 [Router] Original message - Testing  

WARNING file:stream_interface.py __reader line:179 Meshtastic serial port disconnected, disconnecting... read failed: [Errno 6] Device not configured
```

garthvh, 2023-02-02:
Does this happen when you send a message in using JSON?

Kollensperger, 2023-02-02:
Yes, but it also seems to happen on its own after a while, without me actively sending a message - I brought this up on discord but at that time I had trouble connecting via serial because it would crash before I had issued the commands, so it seemed like serial was not working.

I will do a run now without any JSON msg to confirm and see what the output is then

Kollensperger, 2023-02-02:
There are JSON msgs incoming even when I am not actively sending one so that seems to be a possible cause:

```
INFO  | 18:30:29 34 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | 18:30:29 34 [NodeInfoModule] sending owner !52844599/4G2/4599
DEBUG | 18:30:29 34 [NodeInfoModule] Update DB node 0x52844599, rx_time=1675362629
DEBUG | 18:30:29 34 [NodeInfoModule] handleReceived(LOCAL) (id=0x5cd494f9 fr=0x88 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1675362629 priority=10)
DEBUG | 18:30:29 34 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | 18:30:29 34 [NodeInfoModule] Add packet record (id=0x5cd494f9 fr=0x88 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1675362629 priority=10)
INFO  | 18:30:29 34 [NodeInfoModule] Should encrypt MQTT?: 0
DEBUG | 18:30:29 34 [NodeInfoModule] publish msh/2/c/Home/!52844599, 88 bytes
INFO  | 18:30:29 34 [NodeInfoModule] serialized json message: {"channel":0,"from":1384400264,"id":1557435641,"payload":{"hardware":9,"id":"!52844599","longname":"4G2","shortname":"4599"},"sender"
```

and there it crashes again

caveman99, 2023-02-03:
`INFO  | ??:??:?? 25 [Router] JSON publish message to `W , 134 bytes: {"channel":0,"from":-1345136045,"id"`

I assume that W should read 'Weather' - there's something odd with the channel name conversion.

caveman99, 2023-02-03:
the Arduino String() class is known to cause this kind of behaviour, depending on implementation. I have prepared a possible fix for this in https://github.com/meshtastic/firmware/pull/2246 - as soon as it is done building you can download a test firmware from that webpage. Can you please verify if that fixes it?

Kollensperger, 2023-02-04:
Hi, it still fails in the same place, left the other, non ethernet node off for a while, but the moment you switch that one on and the ethernet enabled RAk receives a packet, it crashes:

```
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 2 Filesystem files:
DEBUG | ??:??:?? 2  channels.proto (252 Bytes)
DEBUG | ??:??:?? 2  config.proto (75 Bytes)
DEBUG | ??:??:?? 2  module.proto (63 Bytes)
DEBUG | ??:??:?? 2  db.proto (314 Bytes)
DEBUG | ??:??:?? 2 Using analog input 5 for battery level
INFO  | ??:??:?? 2 No I2C devices found
INFO  | ??:??:?? 2 Meshtastic hwvendor=9, swver=2.0.18.e4843c0
DEBUG | ??:??:?? 2 Reset reason: 0x0
DEBUG | ??:??:?? 3 Setting random seed 4096553454
INFO  | ??:??:?? 3 Initializing NodeDB
INFO  | ??:??:?? 3 Loading /prefs/db.proto
INFO  | ??:??:?? 3 Loaded saved devicestate version 20
INFO  | ??:??:?? 3 Loading /prefs/config.proto
INFO  | ??:??:?? 3 Loaded saved config version 20
INFO  | ??:??:?? 3 Loading /prefs/module.proto
INFO  | ??:??:?? 3 Loaded saved moduleConfig version 20
INFO  | ??:??:?? 3 Loading /prefs/channels.proto
INFO  | ??:??:?? 3 Loaded saved channelFile version 20
INFO  | ??:??:?? 3 No /oem/oem.proto preferences found
INFO  | ??:??:?? 3 Wanted region 3, using EU_868
DEBUG | ??:??:?? 3 region=3, NODENUM=0xf7160fca, dbsize=3
DEBUG | ??:??:?? 3 Using MSL altitude model
DEBUG | ??:??:?? 3 WANT GPS=1
WARN  | ??:??:?? 4 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 4 GxGSA NOT available
INFO  | ??:??:?? 4 External Notification Module Disabled
WARN  | ??:??:?? 4 SX126xInterface(cs=42, irq=47, rst=38, busy=46)
INFO  | ??:??:?? 4 Starting meshradio init...
INFO  | ??:??:?? 4 Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 4 Set radio: region=EU_868, name=Home, config=0, ch=0, power=27
INFO  | ??:??:?? 4 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
INFO  | ??:??:?? 4 Radio myRegion->numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 4 Radio channel_num: 0
INFO  | ??:??:?? 4 Radio frequency: 869.525024
INFO  | ??:??:?? 4 Slot time: 42 msec
INFO  | ??:??:?? 4 Set radio: final power level=22
INFO  | ??:??:?? 4 SX126x init result 0
INFO  | ??:??:?? 4 Frequency set to 869.525024
INFO  | ??:??:?? 4 Bandwidth set to 250.000000
INFO  | ??:??:?? 4 Power output set to 22
DEBUG | ??:??:?? 4 Current limit set to 140.000000
DEBUG | ??:??:?? 4 Current limit set result 0
INFO  | ??:??:?? 4 Set Rx Power Saving Gain mode; result: 0
INFO  | ??:??:?? 4 SX1262 Radio init succeeded, using SX1262 radio
INFO  | ??:??:?? 4 starting Ethernet DHCP
INFO  | ??:??:?? 53 Local IP 192.168.8.149
INFO  | ??:??:?? 53 Subnet Mask 255.255.255.0
INFO  | ??:??:?? 53 Gateway IP 192.168.8.1
INFO  | ??:??:?? 53 DNS Server IP 192.168.8.1
DEBUG | ??:??:?? 53 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
DEBUG | ??:??:?? 53 myNodeInfo.bitrate = 74.341286 bytes / sec
INFO  | ??:??:?? 53 PowerFSM init, USB power=1
INFO  | ??:??:?? 53 Enter state: BOOT
DEBUG | ??:??:?? 53 [Power] Battery: usbPower=0, isCharging=0, batMv=4050, batPct=83
INFO  | ??:??:?? 53 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 53 [NodeInfoModule] sending owner !f7160fca/4GRouter/4GR
DEBUG | ??:??:?? 53 [NodeInfoModule] Initial packet id 954635247, numPacketId 4294967295
DEBUG | ??:??:?? 53 [NodeInfoModule] Update DB node 0xf7160fca, rx_time=0
DEBUG | ??:??:?? 53 [NodeInfoModule] handleReceived(LOCAL) (id=0x38e693f1 fr=0xca to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 53 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 53 [NodeInfoModule] Add packet record (id=0x38e693f1 fr=0xca to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
INFO  | ??:??:?? 53 [NodeInfoModule] Should encrypt MQTT?: 0
INFO  | ??:??:?? 53 [NodeInfoModule] MQTT not connected, queueing packet
DEBUG | ??:??:?? 53 [NodeInfoModule] Using AES256 key!
[... 103 lines cut ...]
DEBUG | 06:51:23 126 [Router] Using channel 0 (hash 0x92)
DEBUG | 06:51:23 126 [Router] Using AES256 key!
DEBUG | 06:51:23 126 [Router] Software encrypt fr=afd2da53, num=17204c5, numBytes=18!
DEBUG | 06:51:23 126 [Router] decoded message (id=0x017204c5 fr=0x53 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1675493483 rxSNR=6 rxRSSI=-21)
DEBUG | 06:51:23 126 [Router] handleReceived(REMOTE) (id=0x017204c5 fr=0x53 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1675493483 rxSNR=6 rxRSSI=-21)
DEBUG | 06:51:23 126 [Router] Module 'DeviceTelemetry' wantsPacket=1
INFO  | 06:51:23 126 [Router] Received DeviceTelemetry from=0xafd2da53, id=0x17204c5, portnum=67, payloadlen=14
INFO  | 06:51:23 126 [Router] (Received from PK): air_util_tx=0.000000, channel_utilization=0.000000, battery_level=100, voltage=4.228000
DEBUG | 06:51:23 127 [Router] updateTelemetry REMOTE node=0xafd2da53 
DEBUG | 06:51:23 127 [Router] Node status update: 0 online, 3 total
DEBUG | 06:51:23 127 [Router] Module 'DeviceTelemetry' considered
DEBUG | 06:51:23 127 [Router] Module 'EnvironmentTelemetry' wantsPacket=1
INFO  | 06:51:23 127 [Router] Received EnvironmentTelemetry from=0xafd2da53, id=0x17204c5, portnum=67, payloadlen=14
DEBUG | 06:51:23 127 [Router] Module 'EnvironmentTelemetry' considered
DEBUG | 06:51:23 127 [Router] Module 'routing' wantsPacket=1
INFO  | 06:51:23 127 [Router] Received routing from=0xafd2da53, id=0x17204c5, portnum=67, payloadlen=14
[... cut ...]
INFO  | 06:51:23 127 [Router] serialized json message: {"channel":0,"from":-1345136045,"id":24249541,"payload":{"air_util_tx":0,"battery_level":100,"channel_utilization":0,"voltage":4.2280
INFO  | 06:51:23 127 [Router] JSON publish message to msh/2/json/Home/!f7160fca, 216 bytes: {"channel":0,"from":-1345136045,"id":24249541,"payload":{"air_util_tx":0,"battery_level":10
WARNING file:stream_interface.py __reader line:179 Meshtastic serial port disconnected, disconnecting... read failed: [Errno 6] Device not configured

```

caveman99, 2023-02-04:
the channel display works now, but that was not the underlying problem it seems.

Kollensperger, 2023-02-04:
Built and installed firmware swver=2.0.19.58743dc4 from the bug-2242 branch, which also crashes at the same point - not sure that one was done but I just tried it to see if I could get it to work via VSCode and platform io :)

caveman99, 2023-02-04:
the 2242 branch has the string fix. at least the debug print works now. Unfortunately i need a RAKDAP or similar hardware debugger to trace this further and i don't have it with me for at least another week - maybe if you have something like this you can hook it up to the bottom eyes of the 4631 and trace the crash in VSCode.

Kollensperger, 2023-02-04:
I have a segger and an atmel debugger at work around the corner - If either of those can work I'll see if I can figure this out - is there some hookup info on the RAK website for this?

caveman99, 2023-02-04:
i think segger should work. ref: https://forum.rakwireless.com/t/hint-platformio-segger-jlink-rtos-task-debugging/4498

Pinout is here https://github.com/RAKWireless/WisBlock/tree/master/bootloader/RAK4630#update-over-jlink

Kollensperger, 2023-02-04:
Great - I've got them talking it seems - but since I've never used a debugger before I could do with some help - which of the outputs do you need? There are several breakpoints and items flagged in something called call stack, but no single file I can see that I could forward to you

caveman99, 2023-02-04:
Try setting a breakpoint on the LOG_DEBUG statement JSON publish message to ...

When execution stops, single-trace from there till it crashes and make a screenshot ;-)

Kollensperger, 2023-02-04:
Not sure I've added the breakpoint in the right place, but it at least compiles.
I press "Step Over" until I reach that breakpoint and then " Step Into" repeatedly.
However I am unsure how I in this mode can see that it crashes as it just seems to cycle over and over again, even if the RAK should have crashed at that point.

I have attached a screenshot for your perusal:

[image]

Kollensperger, 2023-02-11:
If anyone has any pointers as to how I can help debug this further then I can give it a try this weekend.

D4rk4, 2023-02-19:
Any progress with this issue?

Kollensperger, 2023-02-20:
> Any progress with this issue?

I'm stuck and hoping @caveman99 can have a look at some point as I am unsure how to debug this further.

Kollensperger, 2023-04-13:
Hi @caveman99 @garthvh
Coming back to this to see if any of the newer versions of the firmware might have a fix for this - I see some mqtt related comments on 2.1.8?

thebentern, 2023-05-25:
Is this still an issue?

Kollensperger, 2023-05-25:
yes its still an issue  - last version I tried is v 2.1.10 
I really hope this can be fixed as I have tried to go the other way and post messages via the protobuf and msh/2/c/secondary channels but its not as userfriendly for non-expert users as JSON.

One or two other users have messaged me to say that they are also experiencing this problem and hoping for a resolution.

garthvh, 2023-05-25:
Contributions are welcome but the JSON mode was advertised as user friendly, but it is very unstable it has been refactored once without success.  Using the native protobufs is recommended, and honestly not that hard think of it like JSON with no brackets.

Kollensperger, 2023-05-25:
I think @caveman99 was going to take another look at it time permitting. JSON has worked great for me apart from this one issue that remains as an obstacle to deploying my setup. Like I say I have found the JSON part easier and have put some effort into messing about with the protobufs. 
A significant amount of work has been put into documenting JSON and building flows based on JSON on the forums as far as I can see.  I've understood @garthvh you're no fan of JSON in meshtastic and I don't have the competence to contradict, but at some point it would make sense to either fix it or drop it, in which case the related links and documentation also should be removed.

geeksville, 2023-06-06:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/here-is-a-newb-mqtt-tutorial-for-1-3/6583/28

thebentern, 2023-09-13:
Closing. I believe this has been resolved with the uint type conversion for the node id in the json payload in 2.2.x

closed 2023-09-13 by thebentern