meshtastic/firmware #2251: [Bug]:

mchuzie, 2023-02-04:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.18.1a7991c

### Description

Upgraded from 16 to 18. Messages timestamps on device are replaced with random characters. Downgraded back to 16 and timestamps are restored.

garthvh, 2023-02-04:
Relative time has replaced the UTC timestamp so that it is not necessary to manage timezones and daylight savings time on device.

caveman99, 2023-02-04:
@garth it looks like sometimes the relative timestamp prints garbage. i want to add a guard clause and check if values are plausible.

tavdog, 2023-10-12:
Has this been fixed somewhere ? I haven't experienced this issue on 2.1

closed 2023-10-13 by thebentern