meshtastic/firmware #2257: [Bug]: Heap-Leak at sending via CLI

karamo, 2023-02-07:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.18

### Description

malicious behavior: When I extend the MT-CLI (python) with the --sendrepeattext function
and e.g. 10x send a message at intervals of 20 seconds.
The heap then drops continuously also after the send ends (1) and remains at a lower heap level after some time (2).
[image]
[image]

caveman99, 2023-02-08:
Sorry, but just sending a message queues stuff up all around. What is the display aftre 15 minutes?

karamo, 2023-02-08:
> Sorry, but just sending a message queues stuff up all around. What is the display aftre 15 minutes?

constant at lower Heap.

karamo, 2023-02-08:
> Sorry, but just sending a message queues stuff up all around. What is the display aftre 15 minutes?

constant at lower heap. see red line.
[image]

thebentern, 2023-02-20:
The CLI has a more transient "PhoneAPI" connection lifecycle than the apps, so I'm not entirely surprised

karamo, 2023-02-22:
Today I tested with FW 2.0.21 + tryfix.
Result: No Heap-Leak
[image]
The up-peaks are the time I sent a message via CLI.

Issue can bee closed.

closed 2023-02-22 by thebentern