meshtastic/firmware #2258: [Bug]: Heap-Leak at sending via APP

karamo, 2023-02-07:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.18

### Description

Look at the **orange line** = TLORA 2.1_1.6 with FW 2.018 + WiFi & MQTT
From a stable Heap ... sending 2 Messages in the Android APP 2.0.16 ... Heap increases and then decreases continuosly.
[image]

Info: magenta & blue are FW 2.0.17
green and ligthgreen are FW 2.0.18
red is FW 2.018 after sending 10 messages via CLI

caveman99, 2023-02-08:
probably duplicate of #2257 - sending a message is the same from CLI as it is from APP

karamo, 2023-02-08:
> probably duplicate of #2257 - sending a message is the same from CLI as it is from APP

No, it's not the same. Here the Heap is falling down ... Heap-Reboot.

karamo, 2023-02-22:
### **Need HELP !!!**
With Android APP 2.0.20 I can't connect to a node via IP with FW 2.0.21.

karamo, 2023-03-16:
With FW 2.1.2 + Patch https://github.com/meshtastic/firmware/pull/2359 Heap ist ok.
No longer Heap-Leak.

Issue can bee closed.

closed 2023-03-16 by thebentern