meshtastic/firmware #2259: [Feature Request]: turn off Wi-Fi below a certain battery level

IZ1IVA, 2023-02-07:
### Platform

ESP32

### Description

For some use cases it might be useful to make power-hungry ESP32 devices reduce their energy budget by automatically turning off Wi-Fi as battery level falls below a certain (configurable?) value **v**.

- Priority would then be given to "normal" Meshtastic connectivity, i.e. LoRa.
- This way, battery runtime could be almost doubled for the balance of the battery capacity.
- Wi-Fi could be turned back on when battery level climbs back to **v** + margin.

E.g.: turn off below 20%, turn back on at 20% + 10% margin= 30% battery level

IZ1IVA, 2023-02-07:
> Is there much battery powered Wi-Fi use?

I can't speak for others, but I do use Wi-Fi on some battery/solar-battery powered nodes. My use case is mainly fixed nodes on difficult to access roofs, but close enough to use Wi-Fi for MQTT and/or web access.

megabyte2010, 2023-02-08:
I can add another request to this. Option to set wifi power. Sometimes we don´t need wifi full power because the router is near. We can save some battery power with that setting.

IZ1IVA, 2023-02-09:
> I can add another request to this. Option to set wifi power. Sometimes we don´t need wifi full power because the router is near. We can save some battery power with that setting.

IZ1IVA, 2023-02-09:
> I can add another request to this. Option to set wifi power. Sometimes we don´t need wifi full power because the router is near. We can save some battery power with that setting.

Good idea. Perhaps you could open a separate feature request to maximise its visibility.

RightHandMan1, 2023-04-15:
IMHO, I would think if something like this is wanted, it would be as an option, true/false. My concern is, the Heltec V3s(maybe other 32S3, already have incorrect battery levels when down from 4.2v. Even if ADC is matched at 4.2v, at batt 3.9v, the firmware thinks its 2.9v(or less) . Thus if this enhancement/option is enabled, it would further decrease the use time of the battery. Not all of us have the mqtt wifi node on usb or solar.

garthvh, 2023-04-17:
Seems likely to be really inaccurate on every esp32 model other than a tbeam.

IZ1IVA, 2023-04-17:
> Seems likely to be really inaccurate on every esp32 model other than a tbeam.

Can't you just fiddle with the ADC multiplier override? I did that with all my ESP32 devices to make them report the correct voltage. T-LoRas and T-Beams both need a value around 2.21 in my tests.

closed 2023-12-02 by thebentern