meshtastic/firmware #2264: [Bug]: Reboot Loop heap too low

daniel-hofer, 2023-02-07:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.18.1a7991c

### Description

T-Beam with set up to WIFI and MQTT default activated and admin Channel set reboots with the line
====== heap too low [11000] -> reboot in 1s ======
[heaptolowtbeam2.0.18.log](https://github.com/meshtastic/firmware/files/10680356/heaptolowtbeam2.0.18.log)

### Relevant log output

```Shell
DEBUG | 23:40:18 52 [Blink]

====== heap too low [11000] -> reboot in 1s ======

DEBUG | 23:40:18 52 [Screen] showing reboot screen
INFO  | 23:40:18 52 [mqtt] Received MQTT topic msh/2/c/LongFast/!f71f172c, len=110
DEBUG | 23:40:18 52 [mqtt] Using channel 0 (hash 0x8)
DEBUG | 23:40:18 52 [mqtt] Expanding short PSK #1
DEBUG | 23:40:18 52 [mqtt] Using AES128 key!
DEBUG | 23:40:18 52 [mqtt] ESP32 crypt fr=91a1ee9c, num=49cb21f8, numBytes=59!
DEBUG | 23:40:18 52 [mqtt] decoded message (id=0x49cb21f8 fr=0x9c to=0x60, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:18 52 [Blink]

====== heap too low [11000] -> reboot in 1s ======

DEBUG | 23:40:18 52 [Router] Found existing packet record for fr=0x91a1ee9c,to=0x919ddb60,id=0x49cb21f8
DEBUG | 23:40:18 52 [Router] recentPackets size=131
DEBUG | 23:40:18 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:18 52 [Router] Found existing packet record for fr=0x91a1ee9c,to=0x919ddb60,id=0x49cb21f8
DEBUG | 23:40:18 52 [Router] Add packet record (id=0x49cb21f8 fr=0x9c to=0x60, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:18 52 [Router] recentPackets size=131
DEBUG | 23:40:18 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:18 52 [Router] Ignoring incoming msg, because we've already seen it (id=0x49cb21f8 fr=0x9c to=0x60, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:18 52 [Router] Incoming message was filtered 0x91a1ee9c
DEBUG | 23:40:18 52 [Screen] showing reboot screen
INFO  | 23:40:18 52 [mqtt] Ignoring downlink message we originally sent.
INFO  | 23:40:18 52 [mqtt] Received MQTT topic msh/2/c/LongFast/!250477c8, len=115
DEBUG | 23:40:18 52 [Router] Found existing packet record for fr=0x7efebf34,to=0x919ddb60,id=0x16fba0ee
DEBUG | 23:40:18 52 [Router] recentPackets size=131
DEBUG | 23:40:18 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:18 52 [Router] Found existing packet record for fr=0x7efebf34,to=0x919ddb60,id=0x16fba0ee
DEBUG | 23:40:18 52 [Router] Add packet record (id=0x16fba0ee fr=0x34 to=0x60, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:19 52 [Router] recentPackets size=131
DEBUG | 23:40:19 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:19 52 [Router] Ignoring incoming msg, because we've already seen it (id=0x16fba0ee fr=0x34 to=0x60, WantAck=0, HopLim=4 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:19 52 [Router] Incoming message was filtered 0x7efebf34
INFO  | 23:40:19 52 [mqtt] Ignoring downlink message we originally sent.
INFO  | 23:40:19 52 [mqtt] Received MQTT topic msh/2/c/LongFast/!a3fb6f60, len=95
INFO  | 23:40:19 52 [mqtt] Received MQTT topic msh/2/c/LongFast/!250477c8, len=110
DEBUG | 23:40:19 52 [Router] Found existing packet record for fr=0xa3fb6054,to=0x919ddb60,id=0x6b5e019c
DEBUG | 23:40:19 52 [Router] recentPackets size=131
DEBUG | 23:40:19 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:19 52 [Router] Found existing packet record for fr=0xa3fb6054,to=0x919ddb60,id=0x6b5e019c
DEBUG | 23:40:19 52 [Router] Add packet record (id=0x6b5e019c fr=0x54 to=0x60, WantAck=0, HopLim=1 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:19 52 [Router] recentPackets size=131
DEBUG | 23:40:19 52 [Router] recentPackets size=131 (after clearing expired packets)
DEBUG | 23:40:19 52 [Router] Ignoring incoming msg, because we've already seen it (id=0x6b5e019c fr=0x54 to=0x60, WantAck=0, HopLim=1 Ch=0x0 Portnum=4 requestId=289c42d5 rxtime=1675813193 priority=70)
DEBUG | 23:40:19 52 [Router] Incoming message was filtered 0xa3fb6054
INFO  | 23:40:19 52 [mqtt] Received MQTT topic msh/2/c/LongFast/!a3fb6054, len=112
[... 86 lines cut ...]
DEBUG | 23:40:19 53 [Router] Incoming message was filtered 0x91a1ee9c
INFO  | 23:40:19 53 Rebooting
[ 53100][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 53103][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 23:40:19 53 WiFi-Event 5: Disconnected from WiFi access point
[ 53121][E][WiFiSTA.cpp:353] disconnect(): clear config failed!
[ 53121][E][WiFiSTA.cpp:357] disconnect(): disconnect failed!
Wets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
E (813) esp_core_dump_flash: No core dump p▒.W▒▒▒▒▒found!
E (813) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C
```

caveman99, 2023-02-08:
Can you send your configuration? (meshtastic --info) This line means you would have run out of memory in a few minutes anyway.

rpsainio, 2023-02-08:
It seems that it rebooted after running 52 seconds. What happened before ?

daniel-hofer, 2023-02-08:
I checked the node today after work and figured out, that it rebooted again, think for the most time. I disabled MQTT with the over IP connected Android Phone and after that, no reboots! I activated MQTT now and see no problems at the moment.

daniel-hofer, 2023-02-08:
[meshtasticinfo.txt](https://github.com/meshtastic/firmware/files/10688840/meshtasticinfo.txt)

daniel-hofer, 2023-02-08:
> It seems that it rebooted after running 52 seconds. What happened before ?

It rotates the screen and seemed like in good condition, but after a while, the screen showed rebooting...

caveman99, 2023-02-09:
Your node db is huge. Can you try clearing it from the app? If this is a public relay I suggest enabling repeater mode as that won't save the nodedb

daniel-hofer, 2023-02-09:
I use this node at home to link other nodes via default meshtastic mqtt, so all in our mesh can access them.
And at home I use it to connect my phone to the ip-address of this node to read and send messages. So if you suggest, it would be better to change to repeater mode, I'll try.
Isn't the node-db shared between all nodes in the network anyway? So default mqtt is not a good idea because of the big node db?

caveman99, 2023-02-09:
> Isn't the node-db shared between all nodes in the network anyway?

The NodeDB is a local representation of the stations a node can talk to. It's not neccesarily the same db on all connected nodes.

> So default mqtt is not a good idea because of the big node db?

Exactly. The MQTT was meant as a possibility to bridge private networks (PSK changed) without too much setup work. While this works with the public default channel too, your nodedb will be saturated fast.

Having said that, the size limit of 80 nodes may be problematic for applications where we use other memory intensive processing. I'll add a check to the node allocation to see if we actually have heap left.

daniel-hofer, 2023-02-09:
[meshtasticlog.txt](https://github.com/meshtastic/firmware/files/10695304/meshtasticlog.txt)
Hi, I  made another log with some reboots happened at the moment.

daniel-hofer, 2023-02-09:
[meshtasticinfo.txt](https://github.com/meshtastic/firmware/files/10695587/meshtasticinfo.txt)

daniel-hofer, 2023-02-09:
> The NodeDB is a local representation of the stations a node can talk to. It's not neccesarily the same db on all connected nodes.
>
>
> Exactly. The MQTT was meant as a possibility to bridge private [...]

Thank you for the information!

closed 2023-04-15 by caveman99