meshtastic/firmware #2270: [Bug]: Message queue issue

seydunatirej, 2023-02-10:
### Category

BLE

### Hardware

T-Lora v2 1.6

### Firmware Version

T3V1.6.1

### Description

I’m using Meshtastic 2.0, and I have 2 modules of T3 V1.6.1
I’m doing some experiments with the application and I noticed this;
The messages I send doesn’t goes in order (also one after one), and most of them doesn’t haven’t been sent. Also the messages I’m sending is not going in regular order - it’s going partly. I need to notice that I send the messages one after the other, but the cloud is crossed out (like it haven’t send the message). I wonder how the message queue is, why is it skipping? Maybe we don’t send messages over and over under normal conditions, but I want to understand this issue. I

garthvh, 2023-02-10:
When the cloud is crossed out it means that the message was not acknowledged by another node on the network.  You should be able to look in the android debug log and see what the error was.

seydunatirej, 2023-02-10:
what about the message queue why my messages don't follow the normal format when they are sent fast

GUVWAF, 2023-02-10:
Which exact 2.0 firmware version are you using? PR #1820 added better queue management, which is in the firmware from 2.0.13 on. You also have to use at least version 2.0.13 of the Android app.

seydunatirej, 2023-02-10:
2.0.11 firmware

GUVWAF, 2023-02-10:
Can you please try whether the order is correct using the latest release for firmware and Android? 
Not receiving acknowledgments (cloud crossed out) when sending multiple packets quickly after each other is a separate issue, although LoRa is in general not well-suited for this.

seydunatirej, 2023-02-10:
I tried as you said but the result is the same. I know that under normal circumstances, no one sends messages that fast, but I don't understand why there is some kind of reason.

GUVWAF, 2023-02-10:
It probably happens when retransmissions are needed. Since we don't know exactly when the acknowledgment should have been received, there is a timer set for the retransmission. If you in the meantime try to send another packet, it might be send before the retransmitted packet, resulting in out-of-order delivery. This is a normal phenomenon in computer communication, and TCP for example ensures that eventually packets arrive in order using sequence numbers, but there is no TCP layer on top of LoRa in Meshtastic.

seydunatirej, 2023-02-11:
Applying a check with tcp will reduce the normal speed. Is this why it is not preferred?

GUVWAF, 2023-02-11:
Yes, TCP would add a lot of overhead and is not really suited for Meshtastic. Having sequence numbers is normally also not so important, as we don’t send images or files and if you don’t send too quickly, the packets will arrive in order.

closed 2023-02-11 by garthvh