meshtastic/firmware #2280: [Board]: Nano G1 Explorer

neilhao, 2023-02-16:
### SOC

ESP32

### Lora IC

SX1262

### Product Link

https://wiki.uniteng.com/en/meshtastic/nano-g1-explorer

### Description

The Nano G1 Explorer features a new internal wideband Lora antenna, which supports frequencies from 815 Mhz to 940 Mhz. The wideband antenna, combined with an optimized wideband Lora RF frontend circuit, enables the Nano G1 Explorer to work with the majority of regions' LoRa frequency bands around the world without the need for antenna changes. The design of the device also takes into account the potential effect of the human body on its antenna performance, ensuring optimal RF performance even when carried in a pocket.

**RF Upgrades:**
1. Nano G1 Explorer features a new internal wideband Lora antenna with supports for frequencies from 815 Mhz to 940 Mhz. The optimized wideband Lora RF frontend circuit provides the ability to cover majority of regions in the world without changing the antenna.
2. The device has a new internal GPS antenna that significantly reduces GPS lock time compared to the Nano G1.
3. Lora transceiver has been upgraded from SX1276 to SX1262, resulting in an increase in the maximum TX power from 20 dBm to 22 dBm.
4. SX1262 circuit includes a TCXO (+-1.5 ppm) for improved frequency accuracy.
5. GPS module can be set to a low-power mode with the GPS physical switch.

**Other Upgrades:**
1. Enhanced message notification circuit with the option of LED or Buzzer, configurable through a physical switch.
2. Physical power switch for easy turn on/off of the system.
3. Replaceable OLED screen with a FPC connector.
4. Internal Li-Polymer Battery Charger , an optional 3.7V 603450 JST 1.25mm Rechargeable Li-Polymer Battery could be installed.
5. Buck-boost converter, could provide stable 3.3V to the system even the Li-Polymer battery voltage drops as low as 2.5V. Supply voltage stability is very important to ensure that the performance of RF circuits meets design expectations.

**Typical RF Performance:**
<html><body>

Item | Description
-- | --
Lora Operation Frequency | 815 Mhz to 940 Mhz
Lora Max RF Output Power | 22 dBm
VSWR of Lora PCB Antenna | < =1.5 @ 915 MHz and  868Mhz
Gain of Lora PCB Antenna | 2.0 dBi, Omni-directional
Impedance Bandwidth of Lora PCB Antenna | 815 Mhz to 940 Mhz for VSWR < = 2
Polarization of Lora PCB Antenna | Linear
Positioning And Navigation Module | Build-in GPS Module, LNA and Omni-directional Antenna. GPS and BDS are supported
VSWR of GPS PCB Antenna | < =2 @ 1575.42 MHz, GPS L1 Band
Impedance Bandwidth of GPS PCB Antenna | 1474 Mhz to 1590 Mhz for VSWR <= 2

</body>
</html>

garthvh, 2023-02-16:
Looks awesome Neil!

thebentern, 2023-05-23:
Closing as this is complete now.